dtcooper/raspotify #505: new Raspotify , no sound on HDMI

andiandi69, 2022-01-25:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

I installed Raspotify on Pi Zero W with only HDMI Cable Output.
But there is no Sound.
On Speaker-Test there is no such Device

JasonLG1979, 2022-01-26:
There's a [section in the wiki](https://github.com/dtcooper/raspotify/wiki/Troubleshooting) that will walk you though how to configure your default sound device for your Pi. 
Look under the part that says:
> *My Raspberry Pi does not use my USB sound card!*

I know you're trying to get sound to work over HDMI but the steps are same.

closed 2022-01-26 by JasonLG1979

andiandi69, 2022-01-28:
Hi,

 

i do these steps. , but there’s no sound.

It’s an PiZero W 2 and connected via HDMI to an AVR which is on.

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

JasonLG1979, 2022-01-29:
I would look into this thread:

https://forums.raspberrypi.com/viewtopic.php?t=313124

This is not a Raspotify specific issue, but rather a general audio configuration issue.