dtcooper/raspotify #511: Can't update or install raspotify

Kachelkaiser, 2022-02-02:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

Hi,

I run rasportify for some days.

Now I wanted to update to the latest version. But this is not possible via sudo apt-get upgrade or with the installation script

I get the following error:

> sudo apt install raspotify

> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Some packages could not be installed. This may mean that you have
> requested an impossible situation or if you are using the unstable
> distribution that some required packages have not yet been created
> or been moved out of Incoming.
> The following information may help to resolve the situation:
> 
> The following packages have unmet dependencies:
>  raspotify : Depends: libasound2 (>= 1.2.4)
>              Depends: systemd (>= 247.3) but 241-7~deb10u8+rpi1 is to be installed
>              Depends: init-system-helpers (>= 1.60) but 1.56+nmu1 is to be installed
> E: Unable to correct problems, you have held broken packages.

How can I reinstall raspotify?

Gringles, 2022-02-02:
Raspotify recently updated to 0.31.8.1 which added minimum dependencies. This error probably means your OS isn't compatible with Raspotify (e.g. running debian buster when you need debian bullseye). If raspotify was working for you before, try manually downloading 0.31.8 and installing that (it wont throw any errors installing, but might not work once installed).

JasonLG1979, 2022-02-02:
@Gringles hit the nail on the head.

closed 2022-02-02 by github-actions[bot]