dtcooper/raspotify #516: audio device: usb output not working anymore

nikogusm, 2022-02-09:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

Hello,

i was using a previous version of Raspotify on my RaspberryOS10 correctly, outputing the audio to my usb sound card.

After upgrading to Bullseye (os11) also Raspotify is updated and now there are many settings that are different from the previous. I checked it all: now there is also a setting to configure the audio output device.

i tried to set up my setting as the previous one: I set up the default audio output to USB sound card from the alsamixer default config file, but it doesn’t work.

i also tried to set it up on the raspotify configuration file following the documentation but nothing.

now, when i select on Spotify my raspberry as output device it tries to connect to it and after some seconds it disconnects. if i set up raspotify with the default config it works but no sound (because output device is not set up correctly).

the only thing that i’m not sure is that in the config file of raspotify you tell to open “librespot —device ?” but i don’t know exactly what information it needs in the configuration file to choose the correct device.

waiting for a feedback, thanks

JasonLG1979, 2022-02-09:
What are the outputs of:

aplay -l
aplay -L

jto, 2022-02-12:
Hey there. I add the exact same issue after upgrading to bullseye. 
Here's what solved it for me:

Run `librespot --device ?  | less` and find your USB device in the list. 

Once you found the device name, add it in `/etc/raspotify/conf`. 
For example:
```
LIBRESPOT_DEVICE="sysdefault:CARD=A7"
```

restart raspotify with

```sh
sudo systemctl restart raspotify.service
```

Hope it helps :)

JasonLG1979, 2022-02-12:
> Run `librespot --device ? | less` and find your USB device in the list.

Thanks!!!

That was my next step. In some rare cases (because of an ALSA bug) `librespot --device ?` might not show your device. For bonus points you can also use the output of `librespot --device ?` to tell what format(s) your devices supports and configure `librespot` accordingly. (use the highest possible).

nikogusm, 2022-02-12:
thanks guys! the first time I tried to add my usb device on che raspotify conf i typed the wrong text. but now i have this problem: every time i select raspotify device on Spotify, it disconnects immediately. i don’t know if there is a issue in the configuration file or where. is there a fast way to re initialize raspotify configuration to the original setup so i try to te configure it?

nikogusm, 2022-02-12:
I forgot to say that now I found my usb device on librespot and added it correctly to the raspotify config but nothing… it disconnects after i select the device on Spotify

JasonLG1979, 2022-02-12:
> every time i select raspotify device on Spotify, it disconnects immediately. i don’t know if there is a issue in the configuration file or where.

> I forgot to say that now I found my usb device on librespot and added it correctly to the raspotify config but nothing… it disconnects after i select the device on Spotify

If you're certain it's the correct device it sounds like it's already being used. Only one thing at a time can directly access a soundcard. If you need more than one thing to make noise at a time you'll need to run though a software mixer (dmix, pulseaudio).  

> is there a fast way to re initialize raspotify configuration to the original setup so i try to te configure it?

Sure just copy and paste the config from the wiki.

nikogusm, 2022-02-14:
okay, i set up the default config but i have the same issue: when i select the device on Spotify, it connects and disconnects in some seconds. What can i do? is there a log file where errors are shown?

JasonLG1979, 2022-02-15:
> is there a log file where errors are shown?

You can output the systemd log, there is no log file per se.

```
journalctl -u raspotify -b
```

nikogusm, 2022-02-26:
> Feb 26 05:00:24 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
> Feb 26 05:00:24 raspberrypi librespot[508]: [2022-02-26T04:00:24Z ERROR librespot] Discovery Error: Setting up dns-sd failed: No such device (os error 19)
> Feb 26 05:00:24 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
> Feb 26 05:00:24 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
> Feb 26 05:00:51 raspberrypi systemd[1]: raspotify.service: Scheduled restart job, restart counter is at 1.
> Feb 26 05:00:51 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Client).
> Feb 26 05:00:51 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
> Feb 26 05:01:06 raspberrypi librespot[734]: [2022-02-26T04:01:06Z WARN  libmdns::fsm] couldn't parse packet from 192.168.1.125:5353: type 47 is invalid
> Feb 26 05:01:37 raspberrypi librespot[734]: [2022-02-26T04:01:36Z WARN  libmdns::fsm] couldn't parse packet from 192.168.1.125:5353: type 47 is invalid
> Feb 26 05:02:06 raspberrypi librespot[734]: [2022-02-26T04:02:06Z WARN  libmdns::fsm] couldn't parse packet from 192.168.1.125:5353: type 47 is invalid
> Feb 26 05:02:28 raspberrypi librespot[734]: [2022-02-26T04:02:28Z WARN  libmdns::fsm] couldn't parse packet from 192.168.1.107:5353: query type 65 is invalid
> Feb 26 05:02:28 raspberrypi librespot[734]: [2022-02-26T04:02:28Z WARN  libmdns::fsm] couldn't parse packet from [fe80::1848:2fa6:699:e3dd%2]:5353: query type 65 is invalid

this is the log when I start raspotify. the "no such device" error make me think that I did not configured correctly the device on the config file?

JasonLG1979, 2022-02-27:
> this is the log when I start raspotify. the "no such device" error make me think that I did not configured correctly the device on the config file?

No, that's a mDNS error, in fact all of those are are mDNS errors. It doesn't appear that you network is very mDNS friendly. mDNS is used to facilitate automatic discovery. You disable discovery and use you Spotify credentials but that will make it so that Raspotify will only work with the account who's credentials you've provided.

To do that find in `/etc/raspotify/conf`:

```
# Disable zeroconf discovery mode.
#LIBRESPOT_DISABLE_DISCOVERY=
```

And uncomment like so:

```
# Disable zeroconf discovery mode.
LIBRESPOT_DISABLE_DISCOVERY=
```

Then find:

```
# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
#LIBRESPOT_USERNAME=""

# Password used to sign in with.
#LIBRESPOT_PASSWORD=""
```

and uncomment and provide your Spotify creds:
```
# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
LIBRESPOT_USERNAME="myspotifyusername"

# Password used to sign in with.
LIBRESPOT_PASSWORD="myspotifypassword"
```

nikogusm, 2022-03-04:
I tried to set the configuration file as you told but the problem still remains:

Mar 04 05:05:04 raspberrypi librespot[722]: [2022-03-04T04:05:04Z WARN  libmdns::fsm] couldn't parse packet from 192.168.1.125:5353: type 47 is invalid

I see only this errors in che log file :(

JasonLG1979, 2022-03-04:
You have to restart the Raspotify service for changes in the config to take effect.

nikogusm, 2022-03-12:
okay, now what happens at 18:49:50 is what I get immediately after I play something on Spotify and it disconnects:

Mar 12 18:49:22 raspberrypi systemd[1]: Stopping Raspotify (Spotify Connect Client)...
Mar 12 18:49:22 raspberrypi systemd[1]: raspotify.service: Succeeded.
Mar 12 18:49:22 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Mar 12 18:49:22 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Mar 12 18:49:50 raspberrypi librespot[9512]: ALSA lib pcm_dmix.c:1075:(snd_pcm_dmix_open) unable to open slave
Mar 12 18:49:50 raspberrypi librespot[9512]: [2022-03-12T17:49:50Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'E>
Mar 12 18:49:50 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Mar 12 18:49:50 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Mar 12 18:50:01 raspberrypi systemd[1]: raspotify.service: Scheduled restart job, restart counter is at 1.
Mar 12 18:50:01 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Mar 12 18:50:01 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).

tells that the audio output is already in use? but I don't know by what. is it possible to get to know?

JasonLG1979, 2022-03-12:
Those errors are truncated. When posting terminal output use backticks.

JasonLG1979, 2022-03-12:
Are you using PulseAudio?

nikogusm, 2022-03-12:
sorry, the last part of the log that is missing tells:  error 'ENOENT: No such file or directory' for all the lines.

I tried both using the default configuration (I think pulseaudio is the default one?) and editing the raspotify configuration using alsa.

JasonLG1979, 2022-03-12:
> sorry, the last part of the log that is missing tells: error 'ENOENT: No such file or directory' for all the lines.

That means that the device specified does not exist. Something is misconfigured. 

> I tried both using the default configuration (I think pulseaudio is the default one?) and editing the raspotify configuration using alsa.

What I mean is are you using Raspotify on the desktop version of Raspberry Pi OS or have you installed PulseAudio on a headless version?

Let's start over.

Raspotify defaults to the alsa backend. With the alsa backend if you specify a device directly either in your Raspotify config or via `/etc/asound.conf` no other program can concurrently access the device. To allow for concurrent access you must run though a software mixer like dmix or PulseAudio.

`aplay -L` will tell you your devices.

My output looks like this (only showing the DAC I want to use for brevity's sake):

```
hw:CARD=D10s,DEV=0
    D10s, USB Audio
    Direct hardware device without any conversions
```

`librespot --device ?` will also tell you your devices and also what formats they support.

My output looks like this (again only showing the DAC I want to use for brevity's sake):

```
	------------------------------------------------------

	Device:

		hw:CARD=D10s,DEV=0

	Description:

		D10s, USB Audio
		Direct hardware device without any conversions

	Supported Format(s):

		S32

	------------------------------------------------------
```

The device I want to use is my Topping D10s USB DAC and I want to use it directly since Raspotify(librespot) is the only program that makes noise.

I can do this by either setting `hw:CARD=D10s,DEV=0` as my device in either the `/etc/raspotify/conf` with:

```
# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
LIBRESPOT_DEVICE="hw:CARD=D10s,DEV=0"
```

Or I can set it as my system wide default in `/etc/asound.conf` with:

```
pcm.!default {
    type hw
    card D10s
    device 0
}

ctl.!default {
    type hw
    card D10s
}
```

Again as stated above when directly accessing the device no other program will be able to use it.

JasonLG1979, 2022-03-12:
If you're running a headless Raspberry Pi OS install and you do not need any other PulseAudio feature I do not recommend installing PulseAudio even if you need software mixing. In most cases dmix works just fine and is easier to configure.

nikogusm, 2022-03-14:
Solved! Now it's working, thanks to the log file I discovered what was the problem (librespot_device used was wrong and also the supported format).

Thank you a lot!
Nicola

nikogusm, 2022-03-14:
Another question about mDNS feature: why with the previous versions of raspotify I never had this problem?

I ask you this because now every time I start Raspotify I receive an email from Spotify telling "new login from Spotify other device" and I wish to disable the login in my Raspotify config for using it. I prefer using as the previous that was working without login data. Can I check something else? I also asked my ISP about mDNS router issues but he told me that nothing changed so it has to work as it was working before.

JasonLG1979, 2022-03-14:
Unless you wish to disable discovery librespot does not need your Spotify creds. You're getting that email because you've put your creds in the config but not enabled cred caching. Raspotify by default disables cred caching because for the vast majority of users it's not necessarily and it avoids having an auth token laying around for no reason.

JasonLG1979, 2022-03-14:
Long story short if you want to run without discovery you should also enable cred caching.

To do that comment out the line as shown:

This:

```
# Disable caching of credentials.
# Caching of credentials is not necessary so long as
# LIBRESPOT_DISABLE_DISCOVERY is not set.
LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
```

To this:

```
# Disable caching of credentials.
# Caching of credentials is not necessary so long as
# LIBRESPOT_DISABLE_DISCOVERY is not set.
#LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
```

JasonLG1979, 2022-03-14:
mDNS would more be a router issue than an ISP issue. Librespot's mDNS implementation hasn't changed in a very long time. Librespot's implementation isn't the greatest though but router's very in quality in their side of mDNS also. I have one router that librespot does not work with and another that it works fine with so your mileage may vary.

JasonLG1979, 2022-03-14:
I added it to the wiki:

> > *I put my Spotify username and password in /etc/raspotify/conf and now I get a bunch of nag emails from Spotify?*
>
> Unless you have disable zeroconf (mDNS) discovery, librespot does not need your Spotify credentials, zeroconf with take care of authentication.
>
> However if you have disabled discovery for whatever reason and you provided your credentials you will also want to enable credential caching so that librespot will cache the auth token and not have to login every time you connect. Unless you like nag emails ofc?

JasonLG1979, 2022-04-08:
Closing due to inactivity and a lack of response.

closed 2022-04-08 by JasonLG1979