dtcooper/raspotify #517: Raspotify doesen't install on Raspberry Pi Buster

bobrathbone, 2022-02-09:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

Installation on a Raspberry Pi 4 with RPi OS Bullseye

curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
Fails with syntax error

Solution: run with bash
curl -sL https://dtcooper.github.io/raspotify/install.sh | bash

Once the above is cured the installation fails with a apt-transport-https dependency error.
Solution: Run

sudo apt-get install apt-transport-https
The corrected curl program then runs OK

### Relevant log output

```shell
curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
sh: 60: Syntax error: "(" unexpected (expecting "then")
```

bobrathbone, 2022-02-09:
I don't think that the above is a very useful answer especially when I have already supplied the solution.
Would it not be more sensible to put the correct instructions in "The easy way" on the wiki at https://github.com/dtcooper/raspotify
Namely:
sudo apt-get install apt-transport-https
curl -sL https://dtcooper.github.io/raspotify/install.sh | bash

JasonLG1979, 2022-02-09:
I haven't given you a solution?

JasonLG1979, 2022-02-10:
Thanks for the report. As far as I can tell I've fixed the issue.

closed 2022-02-10 by JasonLG1979

bobrathbone, 2022-02-10:
Many thanks for the update. 
The reason for my my comment about the usefulness of your first reply was that the the first update stated I saw was something like: 

"This issue can probably be easily resolved by reading the docs/wiki"
https://github.com/dtcooper/raspotify/labels/read-the-field-manual
Status: Closed

I took this to mean RTFM. The above reply has now disappeared from this post. My apologies if I misunderstood this although I am still confused why such a reply was posted.