dtcooper/raspotify #518: Raspotify running on RPi Bullseye no longer writes track title to the system journal

bobrathbone, 2022-02-09:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

Raspberry Pi 4 running Raspberry Pi OS Bullseye.

The following command used to display track titles 

journalctl --lines 0 --follow _SYSTEMD_UNIT=raspotify.service
This is no longer the case but I am unsure when this changed. 

I am the author of a popular Internet project running on the Raspberry Pi
See http://bobrathbone.com/raspberrypi/pi_internet_radio.html for further information.
I tail the journal and extract the track title and show it on the radio display.
BTW the Artist name has never been displayed in the journal which is a shame.

This problem also existed on Buster until support for Raspotify on Buster was dropped.

### Relevant log output

```shell
journalctl --lines 0 --follow _SYSTEMD_UNIT=raspotify.service
-- Journal begins at Sat 2021-10-30 12:36:50 BST. --
{ Nothing further displayed when the track changes}
```

closed 2022-02-09 by github-actions[bot]

bobrathbone, 2022-02-09:
I don't think that the above is a very useful answer especially when I have already supplied the solution.
Would it not be more sensible to put the correct instructions in "The easy way" on the wiki at https://github.com/dtcooper/raspotify
Namely:
sudo apt-get install apt-transport-https
curl -sL https://dtcooper.github.io/raspotify/install.sh | bash

bobrathbone, 2022-02-09:
https://github.com/dtcooper/raspotify/labels/help-wanted
The above response to this issue was to "read-the-field-manual". I have read the wiki pages but I can not find any references to journalctl or anything relevant to displaying information track title or artist. Is there some other manual or documents other than the wiki at https://github.com/dtcooper/raspotify/wiki
If anyone has any more information on this I would be grateful.

JasonLG1979, 2022-02-09:
The config file has a quite option that's turned on by default. Comment it out.

JasonLG1979, 2022-02-09:
You could also just leave it and write an on event script that echos the artist name and track title.

bobrathbone, 2022-02-09:
Many thanks that did the trick as far as the track title was concerned.
I commented out LIBRESPOT_QUIET in /etc/raspotify/conf 
# Only log warning and error messages.
#LIBRESPOT_QUIET=

Still no sign of the the artist name however. 

I already have a Python class that provides track information
See https://github.com/bobrathbone/piradio6/blob/master/spotify_class.py 

Please close this issue.

JasonLG1979, 2022-02-09:
librespot does not log the artist name. There is nothing that can be done about that. But like I said librespot can be supplied with an on event script/program that will output more metadata than the logs.

JasonLG1979, 2022-02-09:
If the script were to echo that info it would show up in the logs for you to tail.

JasonLG1979, 2022-02-09:
https://github.com/dtcooper/raspotify/discussions/455#discussioncomment-1665069

JasonLG1979, 2022-02-12:
```python3
#!/usr/bin/python3
import os
import json

# PLAYER_EVENT:
# changed, started, stopped,
# playing, paused, preloading, volume_set
print(
    json.dumps(
        {
            'event': os.getenv('PLAYER_EVENT') or '',
            'old_track_id': os.getenv('OLD_TRACK_ID') or '',
            'new_track_id': os.getenv('TRACK_ID') or '',
            'duration_ms': os.getenv('DURATION_MS') or '',
            'position_ms': os.getenv('POSITION_MS') or '',
            'volume': os.getenv('VOLUME') or '',
        }
    )
)
```
`librespot -q -o ./test.py`

```
{"event": "volume_set", "old_track_id": "", "new_track_id": "", "duration_ms": 0, "position_ms": 0, "volume": "58958"}
{"event": "started", "old_track_id": "", "new_track_id": "6KIKRz9eSTXdNsGUnomdtW", "duration_ms": 0, "position_ms": 0, "volume": 0}
{"event": "changed", "old_track_id": "6KIKRz9eSTXdNsGUnomdtW", "new_track_id": "1BKT2I9x4RGKaKqW4up34s", "duration_ms": 0, "position_ms": 0, "volume": 0}
{"event": "playing", "old_track_id": "", "new_track_id": "6KIKRz9eSTXdNsGUnomdtW", "duration_ms": "258687", "position_ms": "281121", "volume": 0}
{"event": "playing", "old_track_id": "", "new_track_id": "1BKT2I9x4RGKaKqW4up34s", "duration_ms": "214440", "position_ms": "0", "volume": 0}
{"event": "preloading", "old_track_id": "", "new_track_id": "3vBv2Tp0kHyBDUBZJr9GXQ", "duration_ms": "", "position_ms": "", "volume": ""}
{"event": "playing", "old_track_id": "", "new_track_id": "3vBv2Tp0kHyBDUBZJr9GXQ", "duration_ms": "261093", "position_ms": "0", "volume": ""}
{"event": "changed", "old_track_id": "1BKT2I9x4RGKaKqW4up34s", "new_track_id": "3vBv2Tp0kHyBDUBZJr9GXQ", "duration_ms": "", "position_ms": "", "volume": ""}
{"event": "preloading", "old_track_id": "", "new_track_id": "0Ed7MeXx64f6OcIuoTRCg1", "duration_ms": "", "position_ms": "", "volume": ""}
{"event": "playing", "old_track_id": "", "new_track_id": "0Ed7MeXx64f6OcIuoTRCg1", "duration_ms": "215200", "position_ms": "0", "volume": ""}
{"event": "changed", "old_track_id": "3vBv2Tp0kHyBDUBZJr9GXQ", "new_track_id": "0Ed7MeXx64f6OcIuoTRCg1", "duration_ms": "", "position_ms": "", "volume": ""}
```