dtcooper/raspotify #520: Raspotify get status activating (auto-restart)

STR3IKERKRYPTON, 2022-02-13:
### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

I installed the newes version of Raspotify on my pi2b. 
but the raspotify.service is in state active(running) for just one sec. Than it gets loaded(auto-restart). After 10 sec the service restarts and it's active for only one sec again. 
I uses this documentation: https://pimylifeup.com/raspberry-pi-spotify/
Does somebody has any idea?

### Relevant log output

```shell
root@raspberrypi:/home/pi# systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Sun 2022-02-13 12:36:54 CET; 6s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 866 ExecStart=/usr/bin/librespot (code=killed, signal=ILL)
   Main PID: 866 (code=killed, signal=ILL)
        CPU: 773ms
root@raspberrypi:/home/pi# systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Sun 2022-02-13 12:36:54 CET; 8s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 866 ExecStart=/usr/bin/librespot (code=killed, signal=ILL)
   Main PID: 866 (code=killed, signal=ILL)
        CPU: 773ms
root@raspberrypi:/home/pi# systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Sun 2022-02-13 12:36:54 CET; 9s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 866 ExecStart=/usr/bin/librespot (code=killed, signal=ILL)
   Main PID: 866 (code=killed, signal=ILL)
        CPU: 773ms
root@raspberrypi:/home/pi# systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-02-13 12:37:04 CET; 848ms ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
   Main PID: 881 (librespot)
      Tasks: 1 (limit: 415)
        CPU: 666ms
     CGroup: /system.slice/raspotify.service
             └─881 /usr/bin/librespot

Feb 13 12:37:04 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
root@raspberrypi:/home/pi# systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Sun 2022-02-13 12:37:05 CET; 1s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 881 ExecStart=/usr/bin/librespot (code=killed, signal=ILL)
   Main PID: 881 (code=killed, signal=ILL)
        CPU: 772ms
root@raspberrypi:/home/pi# systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: signal) since Sun 2022-02-13 12:37:05 CET; 2s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 881 ExecStart=/usr/bin/librespot (code=killed, signal=ILL)
   Main PID: 881 (code=killed, signal=ILL)
        CPU: 772ms
root@raspberrypi:/home/pi#

-------------------------------------------------

root@raspberrypi:/home/pi# systemctl cat raspotify.service
# /lib/systemd/system/raspotify.service
[Unit]
Description=Raspotify (Spotify Connect Client)
Documentation=https://github.com/dtcooper/raspotify
Documentation=https://github.com/librespot-org/librespot
Documentation=https://github.com/dtcooper/raspotify/wiki
Documentation=https://github.com/librespot-org/librespot/wiki/Options
Wants=network.target sound.target
After=network.target sound.target

[Service]
DynamicUser=yes
SupplementaryGroups=audio

Restart=always
RestartSec=10

# Some of these may be implied by DynamicUser=yes
RestrictRealtime=true
RestrictSUIDSGID=true
RestrictNamespaces=true
RestrictAddressFamilies=AF_UNIX AF_INET AF_INET6 AF_NETLINK

ProtectHostname=true
ProtectControlGroups=true
ProtectKernelLogs=true
ProtectKernelModules=true
ProtectKernelTunables=true
ProtectProc=invisible
ProtectHome=true
ProtectSystem=strict
ProtectClock=yes

DevicePolicy=strict
DeviceAllow=char-alsa rw
DeviceAllow=/dev/null r
DeviceAllow=/dev/random r
DeviceAllow=/dev/urandom r

UMask=077
CacheDirectoryMode=0700
CacheDirectory=%N
StateDirectoryMode=0700
StateDirectory=%N
ConfigurationDirectoryMode=0600
ConfigurationDirectory=%N

PrivateTmp=true
PrivateUsers=true

ProcSubset=pid
NoNewPrivileges=true
PermissionsStartOnly=true
LockPersonality=true
MemoryDenyWriteExecute=true
RemoveIPC=true

CapabilityBoundingSet=

SystemCallArchitectures=native
SystemCallFilter=@system-service
SystemCallFilter=~@privileged @resources
SystemCallErrorNumber=EPERM

Environment=LIBRESPOT_NAME="%N (%H)"
Environment=LIBRESPOT_BACKEND="alsa"
Environment=LIBRESPOT_CACHE=%C/%N
Environment=LIBRESPOT_SYSTEM_CACHE=%S/%N

EnvironmentFile=-%E/%N/conf

ExecStart=/usr/bin/librespot

[Install]
WantedBy=multi-user.target
root@raspberrypi:/home/pi#
```

JasonLG1979, 2022-02-13:
That guide looks fine.

Which revision Pi 2? There are 2. One with a Cortex-A7 (ARMv7/armhf/32bit) and one with a Cortex-A53 (ARMv8/AArch64/arm64/64bit)

`signal=ILL` means that a binary is trying to execute an instruction that doesn't match the machine's native architecture.

You would get that error if you installed the armhf package on a Pi 1 or Pi Zero v1.x for example because the raspotify binary only works for ARMv7 (you should not have that problem with a Pi 2) or if you're running the armhf package on a 64 bit system.

The systemd unit file contains `SystemCallArchitectures=native` which will make armhf (32bit) binaries fail on purpose on arm64(64bit) systems to discourage installing 32bit versions on 64bit systems because there is a 64bit version available. If somewhere along the way you've installed the armhf package on a 64bit system you should uninstall it and install the 64bit version.

JasonLG1979, 2022-02-13:
For the sake of completeness, what is the output of:
```
cat /proc/cpuinfo
```

Particularity the hardware and model. Here is for example what my Pi 4 says:

```
Hardware	: BCM2835
Model		: Raspberry Pi 4 Model B Rev 1.4
```

STR3IKERKRYPTON, 2022-02-13:
root@raspberrypi:/home/pi# cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 0010
Serial          : 00000000ffbd49eb
Model           : Raspberry Pi Model B Plus Rev 1.2
root@raspberrypi:/home/pi#

JasonLG1979, 2022-02-13:
> Hardware : BCM2835
> Revision : 0010
> Serial : 00000000ffbd49eb
> Model : Raspberry Pi Model B Plus Rev 1.2
> root@raspberrypi:/home/pi#

That's a Pi 1 not a Pi 2. Raspotify is not compatible with Pi 1's.

closed 2022-02-13 by JasonLG1979

STR3IKERKRYPTON, 2022-02-13:
But I had already installed raspotify on it. I just had to set it up again because I forgot the password.

JasonLG1979, 2022-02-13:
> But I had already installed raspotify on it. I just had to set it up again because I forgot the password.

And in the meantime ARMv6 support was dropped. The last version that was built with ARMv6 support was:

https://github.com/dtcooper/raspotify/releases/tag/0.31.8.1

You can remove the repo and install that version but you will never get updates and bug reports not dealing with configuration will be invalid since you'll be now running an unsupported system with an out of date package. Over time even config related issues may become invalid as things change.