dtcooper/raspotify #522: Unsupported architecture

hushdauihdhi, 2022-03-01:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

I am by no means saying this isn't a stupid error on my end because I am fairly new to this but I am trying to use this software on my Pi Zero W using the lite OS and I am getting the error:
Unspported architecture:

Please make sure you are running a compatible armhf (ARMv7), arm64, or amd64 Debian based OS.

The package installed perfectly fine and I am fairly confident I am on a supported OS but even then, I ran cat /etc/os-release and got:

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

I am sorry if this is clearly answered somewhere but I couldn't find it on the read me and am not very good at navigating GitHub yet.

adrianlyjak, 2022-03-01:
@hushdauihdhi pi zero is ARMv6 which is no longer supported. I'm in the same boat. It is pretty clearly noted in the README.md that it was dropped here: https://github.com/dtcooper/raspotify/commit/345f15c5d695736db8f90d1acc7e542803db5ca0

you can view your system architecture with

```bash
uname -a
# Linux raspberrypi 5.10.63+ #1496 Wed Dec 1 15:57:05 GMT 2021 armv6l GNU/Linux
```

hushdauihdhi, 2022-03-01:
Oh okay, thanks. Do you know of a way I can download an older version? I don't really need the update just a working version will do.

JasonLG1979, 2022-03-01:
> Oh okay, thanks. Do you know of a way I can download an older version? I don't really need the update just a working version will do.

Also in the README.md:

>### [Support for ARMv6 (Pi v1 and Pi Zero v1.x) has been dropped.](https://github.com/dtcooper/raspotify/commit/345f15c5d695736db8f90d1acc7e542803db5ca0)
>
>[0.31.8.1](https://github.com/dtcooper/raspotify/releases/tag/0.31.8.1) was the last version >to be built with ARMv6 support.
>
>*You can install and run that version on an ARMv6 device, but you will never get updates and doing so is completely unsupported.*

closed 2022-03-01 by JasonLG1979