dtcooper/raspotify #523: [Help Needed] Make service work with DNS_SD version of librespot

NePoCz, 2022-03-06:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi, first of all thanks for the great project. I have been using it for a longer time now, but run into a few issues with mDNS lately and so had to compile librespot with DNS-SD (--no-default-features --features "with-dns-sd alsa-backend pulseaudio-backend"). The binary itself works fine, but the service will exit straight after starting, without giving any errors. Does anyone know how to make it work with the dynamic user service that is used in the latest version(s)? 

Here are some more details that might help: 

- The binary works perfectly fine with the 16.0 version service, but does not work with the "new" (31.2+) service using the dynamic user.

- I was able to run the binary with version 31.1, by commenting out the line "RestrictAddressFamilies=AF_INET AF_INET6 AF_NETLINK" in the service file. This service does not yet use dynamic user, but in structure is much more similar to the current service than those done by dtcooper. 

I dont have much knowledge on how does the dynamic user work and I presume the problem is that the binary cant access the avahi daemon API (or complatibility layer?) and hence the service crashes on startup. I was not even able to get any logs out of it. 

Any help would be appreciated

JasonLG1979, 2022-03-06:
This is not a Raspotify issue.

closed 2022-03-06 by JasonLG1979

NePoCz, 2022-03-06:
Hi, I understand this is not an issue, that is why I opened it using your question template - thought that is what this template is for - ask questions or for help. 

This is a (although non-standard) librespot "packaging issue" as mentioned in you readme, so I thought it is relevant. Since you (and many others visiting your repo) surely know much more about systemd dynamic user than me, I was hoping someone migh at least point me in the right direction.  

If you think this is still unrelated, then fine, no problem. I was just hoping this could stay around a bit longe in case anyone has a solution

JasonLG1979, 2022-03-06:
> Hi, I understand this is not an issue, that is why I opened it using your question template - thought that is what this template is for - ask questions or for help.

Questions or help with Raspotify. You've compiled your own binary with different features what you have is no longer Raspotify. 

> This is a (although non-standard) librespot "packaging issue" as mentioned in you readme, so I thought it is relevant.

Raspotify packaging issues. Again, you've compiled your own binary with different features.

> Since you (and many others visiting your repo) surely know much more about systemd dynamic user than me, I was hoping someone migh at least point me in the right direction.

I actually know very little. What I do know was gleaned by trial and error and reading the systemd docs. I followed a very basic and old method for the Raspotify systemd service. I locked it down completely, tried to run it, looked at the logs and gave it *just* enough privileges to run. It was a very long and tedious, iterative process.

> If you think this is still unrelated, then fine, no problem. I was just hoping this could stay around a bit longe in case anyone has a solution

The solution is to read the systemd docs.

jonkerj, 2022-10-20:
While I think I agree with @JasonLG1979 as this is not really a raspotify issue, I found it challenging too to track down the cause and wanted to share my findings on this.

My `librespot` (0.4.2) exits because of DNS-SD: `Could not initialise discovery: Setting up dns-sd failed: DNS-SD Error: Unknown.`. If I wrap `librespot` into `strace -s 200 -f -o /traces/boom` (after adding `/traces` as a `BindPath=` in the unit), I noticed this:

```
33700 connect(14, {sa_family=AF_UNIX, sun_path="/run/dbus/system_bus_socket"}
[..]
33700 geteuid()                         = 61990
[..]
33700 sendto(14, "AUTH EXTERNAL <some number, redacted>\r\n", 26, MSG_NOSIGNAL, NULL, 0) = 26
[..]
33700 read(14, "REJECTED EXTERNAL\r\n", 2048) = 19
```

This tells me D-Bus is willing to play ball, and IIRC is in the call stack between `librespot` and `avahi`. My gut tells me this authentication involves resolving UIDs, and these are temporary and only exist in systemd. When I run `librespot` manually in `strace` (ie, with known UID), the conversation with D-Bus is one where authentication is accepted.

This led me to install `libnss-systemd` (I had not done this, as it's not a hard dependency but just a recommendation), so D-Bus (and others) can actually resolve the temporary UID. It is still not happy (same error), even after a reexec of both systemd and dbus. I think more has to be reloaded, I'm going to hit it with the reboot hammer later.

This all may or may not apply to your situation, but hopefully gives insight on how to debug this (off topic) issue 😉

starypatyk, 2024-06-01:
In case anyone happens to encounter the same issue. Incompatibility between **DynamicUser** and **dns-sd** seems to be an issue in systemd itself. :open_mouth:

See: https://github.com/systemd/systemd/issues/22737 - this comment in particular: https://github.com/systemd/systemd/issues/22737#issuecomment-1107355484