dtcooper/raspotify #529: Unable to install on picroft

archonic, 2022-03-26:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

The Mycroft skill "Play Spotify" is recommending Raspotify to enable connections to Spotify devices so it can do playback. I've installed picroft by downloading their most recent image (v20.08_2020-09-07) and performed updates. Here is the output I get.

```
(.venv) pi@picroft:~ $ sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
curl is already the newest version (7.64.0-4+deb10u2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Unmet minimum required package version(s):

libc6 >= 2.31 is required but 2.28-10+rpt2+rpi1 is installed.
systemd >= 247.3 is required but 241-7~deb10u8+rpi1 is installed.
libasound2 >= 1.2.4 is required but 1.1.8-1+rpt2 is installed.
alsa-utils >= 1.2.4 is required but 1.1.8-2+rpt1 is installed.
libpulse0 >= 14.2 is required but 12.2-4+deb10u1+rpi3 is installed.

Please make sure you are running a compatible armhf (ARMv7), arm64, or amd64 Debian based OS.
```

I'm unsure how to get to the required versions when I'm already on Buster and have performed updates. Do I have to use unstable sources or would it be better to target an older version of Raspotify?

JasonLG1979, 2022-03-26:
Raspotify requires a Debian 11 (the current stable version of Debian at the time of this writing). Buster is Debian 10. You would need to talk to the picroft devs to know when they plan on releasing a Debian 11 based version or possibly how to update to a Debian 11 base.

closed 2022-03-26 by github-actions[bot]

archonic, 2022-03-27:
Thank you for the speedy reply! :pray:

jhthompson12, 2022-03-30:
Hi @archonic , I just ran into this same issue and [this comment](https://github.com/dtcooper/raspotify/issues/511#issuecomment-1027995254) from a similar issue helped me. 

From [this older version](https://github.com/dtcooper/raspotify/releases/tag/0.31.8) I was able to download and install the `..._armhf.deb` on Picroft with a Raspberry Pi 3B and 4.

archonic, 2022-03-30:
@jhthompson12 I ended up building mycroft on Raspbian Bullseye lite x64 (debian 11) using [this recipe](https://github.com/MycroftAI/enclosure-picroft/blob/buster/image_recipe.md) and it went well. The bash steps after running the sh file weren't necessary since they had already been done by the script.