dtcooper/raspotify #531: Sound plays for some time, then suddenly stops

luvxinh, 2022-03-30:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

Headless Raspbian 64 bit. Setup as a worker in Docker swarm node, but currently no container in it.
Setup raspotify using Readme instruction. Connect with a Bose Soundlink via 3.5mm jack.
Raspotify displays fine on list of devices, connects fine. Plays a podcast (I'm using it as whitenoise device for my baby).

What happened: after a while, the audio would just stop. The track seems to be still playing within Spotify app, but there's just no audio.

I happen to notice that if I increase the volume in Spotify app, then the audio would resume. Restarting the raspotify service does not fix it. Reboot the Raspberry pi fixes it as well.

### Relevant log output

```shell
pi@pi42:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2022-03-30 12:20:36 +07; 25min ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
   Main PID: 2639 (librespot)
      Tasks: 2 (limit: 4165)
        CPU: 1min 47.548s
     CGroup: /system.slice/raspotify.service
             └─2639 /usr/bin/librespot

Mar 30 12:20:36 pi42 systemd[1]: Started Raspotify (Spotify Connect Client).
Mar 30 12:20:46 pi42 librespot[2639]: [2022-03-30T05:20:46Z WARN  librespot_connect::spirc] No autoplay_uri found
Mar 30 12:20:46 pi42 librespot[2639]: [2022-03-30T05:20:46Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
Mar 30 12:20:48 pi42 librespot[2639]: [2022-03-30T05:20:48Z WARN  librespot_playback::player] This track may exceed dBFS by 1.22 dB and be subject to 3.22 dB of dynamic limiting at it's pe>
Mar 30 12:21:51 pi42 librespot[2639]: [2022-03-30T05:21:51Z WARN  librespot_connect::spirc] No autoplay_uri found
Mar 30 12:21:51 pi42 librespot[2639]: [2022-03-30T05:21:51Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
Mar 30 12:28:34 pi42 librespot[2639]: [2022-03-30T05:28:34Z WARN  librespot_playback::player] This track may exceed dBFS by 4.49 dB and be subject to 6.49 dB of dynamic limiting at it's pe>
Mar 30 12:39:42 pi42 librespot[2639]: [2022-03-30T05:39:42Z WARN  librespot_connect::spirc] No autoplay_uri found
Mar 30 12:39:42 pi42 librespot[2639]: [2022-03-30T05:39:42Z ERROR librespot_connect::spirc] AutoplayError: MercuryError
Mar 30 12:39:43 pi42 librespot[2639]: [2022-03-30T05:39:43Z WARN  librespot_playback::player] Player::play called from invalid state
```

luvxinh, 2022-03-30:
I do notice that since my set up of Raspotify, my Spotify app does sometime "lost" its status as premium (it displays the tab "premium" at the bottom of the page). Restarting the Spotify app fixes that. Not sure if it's related.

JasonLG1979, 2022-03-30:
Librespot has poor support for podcasts ATM. A fix for that and other things are in the works over there.

luvxinh, 2022-03-31:
Thanks.

I found this issue over at librespot github https://github.com/librespot-org/librespot/issues/894

Do you think I should bring this over there so it'd be easier for discussion? At the moment I have 2 suspicions:

1. due to my Bose Soundlink Revolve. I found this on the Internet. https://[www.markwilson.co.uk](https://www.markwilson.co.uk/blog/2017/07/bose-soundlink-mini-ii-speakers-turn-off-low-volume-levels.htm)/blog/2017/07/bose-soundlink-mini-ii-speakers-turn-off-low-volume-levels.htm
2. due to librespot like you mentioned.

The thing is that I now realize that everytime I lost the sound, if I can raise the volume level to a certain threshold (maybe 70%), then the sound continues, regardless of if it's podcast or song.

JasonLG1979, 2022-04-01:
I would say it's number 1, Overzealous/weird power management that you can't turn off(?). By virtue of that I'm inclined to say this is not our nor librespot's bug.

closed 2022-04-01 by JasonLG1979

luvxinh, 2022-04-02:
Thanks. I've been testing it the past 2 days and it's definitely the 1st issue.

Just want to bump this up to anyone owning a Bose speaker: 
1. You need to use alsamixer to increase the output from the raspberry to 100.
2. Change the initial_volume in raspi config file to 50 (or to whatever level you want). Default is 100 and that would make you unable to increase the volume later using your mobile. Put it to 50 so later you can have some freedom.
3. Change the volume of the speaker (using the hardware button in the speaker) by yourself to an acceptable level.

Hope this helps others too

JasonLG1979, 2022-04-02:
> You need to use alsamixer to increase the output from the raspberry to 100.

I would not even use ALSA volume control ever with that device. I would set it to 100 and never change it.

> Change the initial_volume in raspi config file to 50 (or to whatever level you want). Default is 100 and that would make you unable to increase the volume later using your mobile. Put it to 50 so later you can have some freedom.

No. Raspotify defaults to librespot's default (which is 50) if there is no cached volume and nothing in the config. If the initial volume is set in the config it uses that value, otherwise it uses the cached value.

If you never set a value in the config Raspotify remembers the volume session to session via the cache.

> Change the volume of the speaker (using the hardware button in the speaker) by yourself to an acceptable level.

The problem with the speaker seems to be that if the input gain is not kept high it goes to sleep.

My advice would be:

1. Set the ALSA volume to 100% and never touch it again. (On that I think we basically agree.)

2. Use Raspotify in the way described above to cache the volume level.

3. Use the hardware volume control as much as possible using your controlling device's volume more as a pad so you can keep the input level high to prevent the speaker from going to sleep.