dtcooper/raspotify #532: Interface with Python

Grandpa-G, 2022-03-30:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Is there a way to interface with raspotify from a Python script to start playing a track given a track id as input? This will be run on a Rpi4. I would like to be able to start/stop/pause a song. The OS will be bullseye.

JasonLG1979, 2022-04-01:
No, it's not possible currently to directly control the [librespot](https://github.com/librespot-org/librespot) binary. There are more than few Spotify Web API bindings in about every language. I'm about 99.99% sure there's at least a couple written in Python. I'm pretty sure you can control librespot indirectly though those bindings since librespot just looks like another Spotify Connect device to Spotify.

That's also not to say that a person couldn't write a direct control API for librespot.

JasonLG1979, 2022-04-01:
I'm converting this to a discussion since it's a very broad topic.