dtcooper/raspotify #533: How to actually install older version RPi Zero 1?

fpetru104, 2022-04-01:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hello. Can anyone give me a hint or tell me the command to install the older version of Raspotify still supported by Raspberry Pi Zero? I can't figure it out..

JasonLG1979, 2022-04-01:
To be clear. Your mileage may vary as how well it works and older versions aren't supported so you could end up on your own because it will never be updated. But here you go.

This should get you up and running:

Download the latest version that supports Pi Zero's.
```
wget https://github.com/dtcooper/raspotify/releases/download/0.31.8.1/raspotify_0.31.8.1.librespot.v0.3.1-54-gf4be9bb_armhf.deb
```

Install it.
```
sudo apt install ./raspotify_0.31.8.1.librespot.v0.3.1-54-gf4be9bb_armhf.deb
```

closed 2022-04-01 by JasonLG1979

nkappler, 2022-06-15:
On the website the download command uses curl. This didn't work for me, but wget did work!

MichaIng, 2022-07-05:
With curl it would be:
```sh
curl -LO 'https://github.com/dtcooper/raspotify/releases/download/0.31.8.1/raspotify_0.31.8.1.librespot.v0.3.1-54-gf4be9bb_armhf.deb'
sudo apt install ./raspotify_0.31.8.1.librespot.v0.3.1-54-gf4be9bb_armhf.deb
```