dtcooper/raspotify #536: External USB Soundcard not working any more

old-square-eyes, 2022-04-06:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

On Buster and old Raspotify my USB soundcard used to work. [This is me](https://github.com/dtcooper/raspotify/issues/235#issuecomment-623046360).

`librespot --device ? | less` Shows no devices (I have blacklisted the internal soundcard)

When I try..
`LIBRESPOT_DEVICE="plughw:0,0"` (like used to work from the link above)

I get...
```
Apr 06 18:00:58 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Apr 06 18:01:19 raspberrypi librespot[1283]: [2022-04-06T06:01:19Z ERROR librespot_playback::player] Audio Sink Error Invalid Parameters: <AlsaSink> PCM, ALSA function 'snd_pcm_hw_params' failed with error 'EPIPE: Broken pipe'
Apr 06 18:01:19 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Apr 06 18:01:19 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
```

aplay -l
```
**** List of PLAYBACK Hardware Devices ****
card 0: AudioKontrol1 [Audio Kontrol 1], device 0: Audio Kontrol 1 [Audio Kontrol 1]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
```

aplay -L
```
null
    Discard all samples (playback) or generate zero samples (capture)
hw:CARD=AudioKontrol1,DEV=0
    Audio Kontrol 1, Audio Kontrol 1
    Direct hardware device without any conversions
plughw:CARD=AudioKontrol1,DEV=0
    Audio Kontrol 1, Audio Kontrol 1
    Hardware device with all software conversions
default:CARD=AudioKontrol1
    Audio Kontrol 1, Audio Kontrol 1
    Default Audio Device
sysdefault:CARD=AudioKontrol1
    Audio Kontrol 1, Audio Kontrol 1
    Default Audio Device
dmix:CARD=AudioKontrol1,DEV=0
    Audio Kontrol 1, Audio Kontrol 1
    Direct sample mixing device
```

JasonLG1979, 2022-04-06:
Try:

LIBRESPOT_DEVICE="plughw:CARD=AudioKontrol1,DEV=0"

old-square-eyes, 2022-04-06:
Loud white noise on my first go (after a restart of the service). I hit next track on my phone then lost all audio and Connect disconnected. Tried again and nothing.

```
-- Journal begins at Fri 2022-01-28 15:37:51 NZDT. --
Apr 06 23:18:51 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Apr 07 08:22:39 raspberrypi systemd[1]: Stopping Raspotify (Spotify Connect Client)...
Apr 07 08:22:39 raspberrypi systemd[1]: raspotify.service: Succeeded.
Apr 07 08:22:39 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Apr 07 08:22:39 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Apr 07 08:23:45 raspberrypi librespot[1112]: [2022-04-06T20:23:45Z ERROR librespot_playback::player] Audio Sink Error Invalid Parameters: <AlsaSink> PCM, ALSA function 'snd_pcm_hw_params' failed with error 'EPIPE: Broken pipe'
Apr 07 08:23:45 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Apr 07 08:23:45 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Apr 07 08:23:45 raspberrypi systemd[1]: raspotify.service: Consumed 4.888s CPU time.
Apr 07 08:23:55 raspberrypi systemd[1]: raspotify.service: Scheduled restart job, restart counter is at 1.
Apr 07 08:23:55 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Apr 07 08:23:55 raspberrypi systemd[1]: raspotify.service: Consumed 4.888s CPU time.
Apr 07 08:23:55 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Apr 07 08:24:44 raspberrypi systemd[1]: Stopping Raspotify (Spotify Connect Client)...
Apr 07 08:24:44 raspberrypi systemd[1]: raspotify.service: Succeeded.
Apr 07 08:24:44 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Apr 07 08:24:44 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Apr 07 08:25:08 raspberrypi librespot[1194]: [2022-04-06T20:25:08Z ERROR librespot_playback::player] Audio Sink Error Invalid Parameters: <AlsaSink> PCM, ALSA function 'snd_pcm_hw_params' failed with error 'EPIPE: Broken pipe'
Apr 07 08:25:08 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Apr 07 08:25:08 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Apr 07 08:25:18 raspberrypi systemd[1]: raspotify.service: Scheduled restart job, restart counter is at 1.
Apr 07 08:25:18 raspberrypi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Apr 07 08:25:18 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
```

old-square-eyes, 2022-04-06:
I am at a complete loss. After many hours yesterday, and trying the above today... audio just started working.

The only thing I have done since is downgrade the kernel based on some off Buster sound issues post.

`sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2`

Then rebooted
Then
`sudo rpi-update` back to the latest kernel
Then rebooted

Config I am using is `LIBRESPOT_DEVICE="plughw:0,0"`

I can only imagine something was holding exclusive access to the sound card open and maybe all the rebooting helped.

As a side I also got my Mbox Micro DAC working too. Now I will test Bullseye 64 bit but I'm not hopeful.

JasonLG1979, 2022-04-06:
> The only thing I have done since the above, is downgrade the kernel based on some off Buster old sound issues post.

> sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2

Sounds like a regression in the sound stack below us (ALSA itself).

> Now I will test Bullseye 64 bit but I'm not hopeful. Will report back.

DACs are kinda a mixed bag sometimes. Some people have all kinds of problems, I've got a few DACs, both USBs and Raspberry Pi HATs, and I've never had any problem really with any of them. I don't know if I'm lucky or others are unlucky?

old-square-eyes, 2022-04-07:
64 bit didn't work. Guess that's the kernel driver. I also broke my 64 bit build trying to downgrade to a 32 bit kernel. Clean install with the setup above and it just worked. 32 bit Bullseye.

Guess it's the way you hold your tongue.

Sounds great, hooked up to my hifi, playing my favourite speaker testing D&B playlist reserved for just such an occasion.

Thanks for the great package.

closed 2022-04-08 by JasonLG1979