dtcooper/raspotify #540: Output to pipe?

mrpete69, 2022-04-23:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

I’ve used raspotify for some years combined with snapcast in a multiroom environment, with several rpi+hifiberry beocreate. Now I’m building a new system for one of my kids and don’t understand how to make it work in the new configuration. I’ve googled and tested lots of different ways to get it right, but can’t figure out. Pipe as backend seems obvious, but then I’m stuck. 
In the old raspotify I had this combination of pipe and device: BACKEND_ARGS="--backend pipe --device /tmp/snapfifo"
How should I write this in /etc/raspotify/conf?

JasonLG1979, 2022-04-23:
Duplicate of  https://github.com/dtcooper/raspotify/issues/498

closed 2022-04-23 by JasonLG1979

timglong, 2022-04-24:
@mrpete69 Check out my update on #498, let me know if I can help! Took me 5 minutes or so to get everything working again. Woo!

mrpete69, 2022-04-26:
@timglong Thanks a lot! Worked perfectly!