dtcooper/raspotify #544: Raspotify outputs no sound, and keeps skipping songs

Emilurenius, 2022-05-10:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

I have been using raspotify for two weeks without issues, and then suddenly one day it just stopped working properly. There is no sound output anymore, even though it is outputting to the correct audio device. I have checked that all hardware works, and I am able to get sound out of the raspberry pi with other software. 

Spotify also skips to the next song about 20 times after connecting to the Pi with spotify connect.

I have tried reinstalling the OS and raspotify, but the issue persists.

JasonLG1979, 2022-05-10:
Hardware?
Audio details?
Log?

kirenida, 2022-05-11:
Where in the world are you?

You might have the same issue as I did in #542 

Check your logs and try blocking apresolve.spotify.com if you get ChannelError.

Emilurenius, 2022-05-11:
Here are the logs I am getting: 
`-- Journal begins at Mon 2022-04-04 16:52:30 CEST, ends at Wed 2022-05-11 09:45:58 CEST. --
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z WARN  librespot_playback::player] Skipping to next track,>May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z WARN  librespot_playback::player] Skipping to next track,>May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z WARN  librespot_playback::player] Skipping to next track,>May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z WARN  librespot_playback::player] Skipping to next track,>May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z WARN  librespot_playback::player] Skipping to next track,>May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z WARN  librespot_playback::player] Skipping to next track,>May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z WARN  librespot_playback::player] Skipping to next track,>May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z WARN  librespot_playback::player] Skipping to next track,>May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z WARN  librespot_playback::player] Player::pause called fr>May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z ERROR librespot_playback::player] Unable to load encrypte>May 10 15:38:35 raspi4 librespot[2178]: [2022-05-10T13:38:35Z WARN  librespot_playback::player] Skipping to next track,>lines 1-29...skipping...
-- Journal begins at Mon 2022-04-04 16:52:30 CEST, ends at Wed 2022-05-11 09:45:58 CEST. --
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z WARN  librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 269881097637946755373035848119632447855, audio_typ>
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
May 10 15:38:33 raspi4 librespot[2178]: [2022-05-10T13:38:33Z WARN  librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 124292201033741413902278096745161214135, audio_typ>
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z WARN  librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 276936665107733024281373331087441724011, audio_typ>
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_core::channel] channel error: 2 1
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z WARN  librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 109796569721069273484947060932279210150, audio_typ>
May 10 15:38:34 raspi4 librespot[2178]: [2022-05-10T13:38:34Z ERROR librespot_core::channel] channel error: 2 1
[... cut ...]
May 10 15:38:36 raspi4 librespot[2178]: [2022-05-10T13:38:36Z ERROR librespot_playback::player] Unable to load encrypted file: ChannelError
May 10 15:38:36 raspi4 librespot[2178]: [2022-05-10T13:38:36Z WARN  librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 160693545495108062392491181430724548846, audio_typ>`

I can see why it is skipping songs now. Seems you may be right Kirenida. I am not sure of how to block apresolve.spotify.com. Is it through the firewall on the Pi, or on the router itself?

kirenida, 2022-05-11:
@Emilurenius 

I just put the address in my piHole blacklist, so you can do that if you're using piHole. Just make sure that the pi is set up correctly.

Or you can just add '127.0.0.1 apresolve.spotify.com' to your '/etc/hosts' file on the raspberry.

Or you can do it for your whole network by blocking it through the router.

Emilurenius, 2022-05-11:
> your

That fixed it instantly! Thank you so much!

closed 2022-05-11 by Emilurenius