dtcooper/raspotify #546: Update/install issues on Pi (armv7l GNU/Linux)

rlaferla, 2022-05-20:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

How do I update/install raspotify?  I had it running for a year but now I've encountered some issues and would like to upgrade.   I cloned the GH, installed Docker, and can't get it to work.

```
uname -a
Linux spotify 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
```

```
sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
curl is already the newest version (7.64.0-4+deb10u2).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

Unmet minimum required package version(s):

libc6 >= 2.31 is required but 2.28-10+rpt2+rpi1 is installed.
systemd >= 247.3 is required but 241-7~deb10u8+rpi1 is installed.
libasound2 >= 1.2.4 is required but 1.1.8-1+rpt2 is installed.
alsa-utils >= 1.2.4 is required but 1.1.8-2+rpt1 is installed.
libpulse0 >= 14.2 is required but 12.2-4+deb10u1+rpi3 is installed.

Please make sure you are running a compatible armhf (ARMv7), arm64, or amd64 Debian based OS.
```

JasonLG1979, 2022-05-20:
Raspotify does not support Debian 10. You will need to upgrade to the current stable Debian version.

closed 2022-05-20 by JasonLG1979

JasonLG1979, 2022-05-30:
Dup of https://github.com/dtcooper/raspotify/issues/545