dtcooper/raspotify #548: RaspAp and find raspotify

schorges, 2022-06-22:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

My raspotify running good with the eth0 lan-cable on it and local network i find the raspotify rpi device.

Than i install raspap with nodogsplash and huwai 4G stick, all clients can register an have over eth1 access to the web.

eth0 | ip 192.168.17.86 | gateway 192.168.17.1 <= dhcp router

eth1 | ip 192.168.8.102 | gateway 192.168.8.1

wlan0 | ip 10.3.141.1 | gateway 10.3.141.1

on mobile the raspotiy no found on wlan (not same network)

what do I have to change so that it works without a network cable on eth0 ?

can i change a raspotify ip or the static ip?

tahnks for help !

JasonLG1979, 2022-06-22:
Librespot uses mdns for local discovery.

I know nothing of those other apps. It would be better probably to ask them if it's possible to and how to forward mdns using them.

That being said if your only goal is making librespot accessable to a single user no matter what network they are on you don't need them. All you have to do is provide your creds in the config and enable cred caching.

JasonLG1979, 2022-06-22:
And if you only want 1 user to be able to access Raspotify you can disable discovery.

JasonLG1979, 2022-07-22:
Closing because of a lack of response.

closed 2022-07-22 by JasonLG1979