dtcooper/raspotify #561: Raspberry: upgrade to Bullseye

Mrc527, 2022-07-21:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

I upgraded from buster to bullseye and raspotify started complaining about missing dependencies.

I tried to solve some but it looks like it's a never ending story.

Did you ever try to upgrade from buster? I fear the upgrade is somehow involved.

Please note, libopenmaxil.so should be deprecated in bullseye, if I read correctly. :disa

### Relevant log output

```shell
Jul 21 22:02:17 rpi-Lug systemd[1]: RPiPlay.service: Main process exited, code=exited, status=127/n/a
Jul 21 22:02:17 rpi-Lug systemd[1]: RPiPlay.service: Failed with result 'exit-code'.
Jul 21 22:06:36 rpi-Lug systemd[1]: Started Raspberry PI Airplay service.
Jul 21 22:06:36 rpi-Lug rpiplay[3963]: /usr/local/bin/rpiplay: error while loading shared libraries: libbrcmGLESv2.so: cannot open shared object file: No such file or directory
Jul 21 22:06:36 rpi-Lug systemd[1]: RPiPlay.service: Main process exited, code=exited, status=127/n/a
Jul 21 22:06:36 rpi-Lug systemd[1]: RPiPlay.service: Failed with result 'exit-code'.
Jul 21 22:06:37 rpi-Lug systemd[1]: RPiPlay.service: Scheduled restart job, restart counter is at 572.
Jul 21 22:06:37 rpi-Lug systemd[1]: Stopped Raspberry PI Airplay service.
Jul 21 22:06:37 rpi-Lug systemd[1]: Started Raspberry PI Airplay service.
Jul 21 22:06:37 rpi-Lug rpiplay[3964]: /usr/local/bin/rpiplay: error while loading shared libraries: libbrcmGLESv2.so: cannot open shared object file: No such file or directory

----------

Jul 21 22:02:18 rpi-Lug systemd[1]: RPiPlay.service: Scheduled restart job, restart counter is at 364.
Jul 21 22:02:18 rpi-Lug systemd[1]: Stopped Raspberry PI Airplay service.
Jul 21 22:02:18 rpi-Lug systemd[1]: Started Raspberry PI Airplay service.
Jul 21 22:02:18 rpi-Lug rpiplay[3634]: /usr/local/bin/rpiplay: error while loading shared libraries: libopenmaxil.so: cannot open shared object file: No such file or directory
Jul 21 22:02:18 rpi-Lug systemd[1]: RPiPlay.service: Main process exited, code=exited, status=127/n/a
Jul 21 22:02:18 rpi-Lug systemd[1]: RPiPlay.service: Failed with result 'exit-code'.
```

JasonLG1979, 2022-07-21:
Those logs have nothing to do with Raspotify.

JasonLG1979, 2022-07-21:
That is rpiplay. I had to Google it but apparently it's something to do with airplay?

closed 2022-07-21 by JasonLG1979

Mrc527, 2022-07-21:
Whoops, sorry my fault, I was debugging two things at the same time and I confused those two.

🙈

JasonLG1979, 2022-07-21:
It's all good.