dtcooper/raspotify #565: Why exactly are so many dependencies required, even though librespot works flawlessly without them?

ClaasF, 2022-08-09:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Background for those interested: I'm running an RPi3 with RetroPie and tried to crank that thing up to a full blown living room entertainment hub - including Spotify.

Now, RetroPie is a rather small project and thus not super up to date. Currently they're still on Buster, which ships with pretty outdated libs.

Problem: Raspotify requires a bunch of libraries in (relatively) new versions and thus won't install on Buster.

However, I tried to build librespot directly - which doesn't explicitely require any of these requirements and it worked perfectly fine (although it took forever to build on the RPi).

Did I overlook something here? I skimmed through the code and couldn't find anything where these dependencies are actually used, they're just referenced in the install script. 

And if they're indeed not needed, would it be possible, to include a build target for older dependencies?

Dependencies in question:
[libc6 (>= 2.31)](https://tracker.debian.org/pkg/libc6)
[systemd (>= 247.3)](https://tracker.debian.org/pkg/systemd)
[libasound2 (>= 1.2.4)](https://tracker.debian.org/pkg/libasound2)
[alsa-utils (>= 1.2.4)](https://tracker.debian.org/pkg/alsa-utils)
[libpulse0 (>= 14.2)](https://tracker.debian.org/pkg/libpulse0)
[init-system-helpers (>= 1.60)](https://tracker.debian.org/pkg/init-system-helpers)

JasonLG1979, 2022-08-09:
Those deps are absolutely required for librespot to function or for Raspotify the package to install in the case of int-system-helper. The version numbers were chosen specifically so that Raspotify can not be installed on any Debian version older than Stable. The reason being is that I have no interest or time to support multiple Debian versions.

closed 2022-08-09 by JasonLG1979

JasonLG1979, 2022-08-09:
Libc is a given. Linux itself requires a c runtime.

Systemd is required because Raspotify is a systemd service.

Libasound2 is basically ALSA, without it you have no sound subsystem.

Alsa-utils are ALSA utilities useful for doing things like finding your soundcard and other troubleshooting.

Libpulse0 is required because the librespot binary that ships with Raspotify come with pulseaudio backend, so even if you don't use the pulseaudio backend librespot will crash.

Int-system-helper is as I said for installing the deb package.

JasonLG1979, 2022-08-09:
Most of those should already be installed on something like Retropie or Raspbian or any sort of desktop Debian distro (although package versions may not be up to stuff). But on a vanilla Debian server install many of them are not installed, basically it doesn't come with any audio libs.

JasonLG1979, 2022-08-09:
Debian 11 has been out now for over a year. If I were you I would be wondering why Retropie hasn't updated it's base to 11 yet.

ClaasF, 2022-08-09:
Well, as I said, RetroPie isn't exactly a large project and from my personal playing around, getting all those subprojects, modules and plugins to work together can be a bit tedious.

What I found odd is, that raspotify explicitly sets dependencies without actually needing them in these specific versions. 

As I wrote above, librespot does compile and run perfectly fine on Buster. But setting the versions to deliberately exclude too old Debian releases does make sense.

JasonLG1979, 2022-08-09:
Librespot may work fine compiled on your system against your systems lib versions but there is no guarantee that a version complied against vanilla Debian 11 will work with Raspbian 10 especially the armhf version. (The armhf version is patched)

JasonLG1979, 2022-08-09:
And to reiterate those deps are absolutely needed. Go ahead and uninstall them and watch your system break.

ClaasF, 2022-08-09:
I compiled everything on the Pi, so patches are accounted for.

Though I would not recommend that, I had to create a 1G swapfile and wait 2h or so to get everything built.

ClaasF, 2022-08-09:
> And to reiterate those deps are absolutely needed. Go ahead and uninstall them and watch your system break.

I wasn't implying that the deps are not needed, my question was rather about the *versions*, nothing else.

JasonLG1979, 2022-08-09:
I know you compiled everything on the pi against the lib versions on your system that are older and not guaranteed to be compatible with the lib versions in Debian 11.

LevitatingBusinessMan, 2022-10-21:
Just like @ClaasF I am also limited to using buster. Would be nice to have precompiled packages for that version,

JasonLG1979, 2022-10-21:
> Just like @ClaasF I am also limited to using buster. Would be nice to have precompiled packages for that version.

Not going to happen

ClaasF, 2022-10-21:
> Just like @ClaasF I am also limited to using buster. Would be nice to have precompiled packages for that version.

I wrote a build script for my Pi back then: https://gist.github.com/ClaasF/ca9bde512ad0403cd5dd80fcb9a8615f

Maybe this is helpful to you, it works fine for me so far. However, you'll need at least 1GB of Swap (on my Pi3, that is) and it takes *really* long. 

I just copied it from my scratchpad, so the script might be broken/missing a line. Feel free to blame me if it doesn't work for you.

JasonLG1979, 2022-10-21:
Good deal, there you go. But please remember that you're on your own with it.

LevitatingBusinessMan, 2022-10-24:
> I wrote a build script for my Pi back then: https://gist.github.com/ClaasF/ca9bde512ad0403cd5dd80fcb9a8615f
>
> Maybe this is helpful to you, it works fine for me so far. However, you'll need at least [...]

Like Jason said, you can just cross compile this. I just modified the Dockerfile to use `debian:buster` and then changed the dependencies in `control.debian.tmpl`. No need to wait hours for your Pi to compile all those Rust modules.