dtcooper/raspotify #566: Raspotify (0.31.8.1) shows "Unavailable to control" in Spotify device list

derekantrican, 2022-08-18:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

I have a Raspberry Pi Zero W (v1). I've installed Raspberry Pi OS Lite and used the README instructions to install Raspotify 0.31.8.1. Everything looks great except it's grayed-out and shows "Unavailable to control" in my Spotify devices list.

[image]

If I run `librespot` it will also show up in the devices list and also as "Unavailable to control".

After some Googling I found #385 but I believe this is different (and that fix was before 0.31.8.1). I've also scoured the Wiki for any troubleshooting tips and haven't found any. What else should I try?

JasonLG1979, 2022-08-19:
The [README](https://github.com/dtcooper/raspotify#support-for-armv6-pi-v1-and-pi-zero-v1x-has-been-dropped) clearly states:
>[Support for ARMv6 (Pi v1 and Pi Zero v1.x) has been dropped.](https://github.com/dtcooper/raspotify/commit/345f15c5d695736db8f90d1acc7e542803db5ca0)
[0.31.8.1](https://github.com/dtcooper/raspotify/releases/tag/0.31.8.1) was the last version to be built with ARMv6 support.

>You can install and run that version on an ARMv6 device, but you will never get updates and doing so is completely unsupported.

There's not really much use debugging something that will never get fixed even if you figure out what the issue is. There will never be another ARMv6 compatible Raspotify build.

I just noted that release so PiZv1 owner know which is the most up to date version still built with PiZv1 support.

If it works for you great, if not, sorry. That version is completely unsupported.

closed 2022-08-19 by JasonLG1979

derekantrican, 2022-08-20:
Just wanted to update that my issue was identical to #253 with the problem being that somehow Spotify Connect doesn't work for podcasts and I had a podcast that I was recently listening to and was using for testing