dtcooper/raspotify #568: Failing to play soung on Ubuntu 22.04 Desktop

tmortagne, 2022-08-25:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Latest Version

- [X] I'm running the latest version of Raspotify

### Due Diligence

- [X] I have done my due diligence

### What happened?

Using raspotify 0.42.3~librespot.v0.4.2-6537c44 installed though the Debian repository.

I can connect fine with the Spotify from my phone but as soon I start playing something the connection is lost.

When looking at the log I see the following:
```
août 25 20:06:13 media librespot[130198]: Failed to create secure directory (//.config/pulse): No such file or directory
août 25 20:06:13 media librespot[130198]: ALSA lib pcm_dmix.c:1032:(snd_pcm_dmix_open) unable to open slave
août 25 20:06:13 media librespot[130198]: [2022-08-25T18:06:13Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA fun>
```

I tried `speaker-test -c2 -l1` and it's working fine.

I also tried with pulseaudio but the result is the same with the following log:
```
août 25 20:42:00 media librespot[131953]: [2022-08-25T18:42:00Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
août 25 20:42:05 media librespot[131953]: Failed to create secure directory (//.config/pulse): No such file or directory
août 25 20:42:05 media librespot[131953]: [2022-08-25T18:42:05Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <PulseAudioSink> Connection refused
```

I guess it's probably more a librespo problem than a raspotify one but since I'm using raspotify I'm starting here.

JasonLG1979, 2022-08-25:
You can't use the Raspotify system level service in combination with the PuseAudio user level service that runs by default on Desktops. That's very well known thing.

> I guess it's probably more a librespo problem than a raspotify one but since I'm using raspotify I'm starting here.

Nope. It's user error. Raspotify isn't designed to be used on Desktop systems the README and wiki says as much very plainly.

JasonLG1979, 2022-08-25:
Librespot itself will work just fine if you run it in a terminal as a user or as a user level service.

closed 2022-08-25 by JasonLG1979