dtcooper/raspotify #570: Card number changes with reboot

jmayday, 2022-09-09:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

How to configure **/etc/asound.conf** so that it's always pointing to correct card number?
After reboot card number changes, basically here is my setup:
```
jmayday@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 3: DAC51090305 [ADI-2 DAC (51090305)], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

I want to user card 3, so I created **/etc/asound.conf**:
```
jmayday@raspberrypi:~ $ cat /etc/asound.conf 
defaults.ctl.card 3
defaults.pcm.card 3
defaults.pcm.dmix.rate 768000
defaults.pcm.dmix.format S32_LE
```

Unfortunately after reboot it might happen, that my external DAC will be not a card 3 but card 1. So I have to go to asound.conf and reconfigure it. Is it possible to prevent such situation?

JasonLG1979, 2022-09-09:
Then you want to reference your card by name then not index.

Use `aplay -L` to get your DAC's name.

My DAC is a Topping D10s so the part I care about in my output is:

```
hw:CARD=D10s,DEV=0
    D10s, USB Audio
    Direct hardware device without any conversions
```

My card's name is `D10s`. So I would use that in this:

```
# /etc/asound.conf

pcm.!default {
    type plug
    slave.pcm {
        type dmix
        ipc_key {
            @func refer
            name defaults.pcm.ipc_key
        }
        ipc_gid {
            @func refer
            name defaults.pcm.ipc_gid
        }
        ipc_perm {
            @func refer
            name defaults.pcm.ipc_perm
        }
        slave {
            pcm {
                type hw
                card D10s
                nonblock {
                    @func refer
                    name defaults.pcm.nonblock
                }
            }
            channels 2
            rate 44100
            format S32_LE
        }
        bindings {
            0 0
            1 1
        }
    }
}

ctl.!default {
    type hw
    card D10s
}
```

When you find your cards name just replace `D10s` with whatever it is. As a side note there is absolutely no point to setting your rate to 768000 unless your goal is wasting CPU cycles and degrading the sound quality. Set the rate to the highest sampling rate of the audio that you typically play. I use 44100 personally because that the sampling rate librespot outputs.

JasonLG1979, 2022-09-09:
If Raspotify is the only thing on you computer that will ever be using the DAC and it supports 44100 you can skip `/etc/asound.conf` and tell `librespot` to just always use the DAC directly bypassing all software mixing by editing `/etc/raspotify/conf` and setting `LIBRESPOT_DEVICE=` to your DAC:

Using my example output from above for me that would mean:
```
LIBRESPOT_DEVICE="hw:CARD=D10s,DEV=0"
```

JasonLG1979, 2022-09-17:
OK, I'm closing this as solved. I'm going to assume you figured it out.

jmayday, 2022-09-17:
Thanks! I don't understand structure of /etc/asound.conf you've given but it works (just replaced D10s with my card name).

jmayday, 2022-09-17:
BTW is there a way to generate that file (passing the card name as param)?

JasonLG1979, 2022-09-17:
I'm sure you could write a quick bash script that accepted the card name as an arg and generated the file with cat or echo?

jmayday, 2022-09-17:
I meant script provided with raspotify, so that after installing raspotify on new raspberry installation I can easily set it up (the asound.conf file) without having to remember how the file looks like or coming back to this GitHub issue. I would be nice addition I think.

JasonLG1979, 2022-09-17:
> I meant script provided with raspotify, so that after installing raspotify on new raspberry installation I can easily set it up (the asound.conf file) without having to remember how the file looks like or coming back to this GitHub issue. I would be nice addition I think.

That's beyond the scope of the project. This really isn't a general Linux sound help service. Using Raspotify may require you to roll up your sleeves, get your hands dirty and learn something.

JasonLG1979, 2022-09-17:
If you need or want a script, write a script.

JasonLG1979, 2022-09-17:
Plus it's not a good idea to just clobber `/etc/asound.conf` with a script unless you know what you're doing or at least have some help. Writing an automated script that would suit everyone's needs is basically impossible because there are a million and one different configurations and user needs. It's better if you wrote your own script for your own needs.

JasonLG1979, 2022-09-18:
@jmayday this should do the trick. You'll need to run it with sudo since it's writing a file in `/etc`

```python3
#!/usr/bin/python3
import subprocess
import signal
import sys
import os

signal.signal(signal.SIGTERM, lambda *args, **kwargs: print("\n") or sys.exit(0))
signal.signal(signal.SIGINT, lambda *args, **kwargs: print("\n") or sys.exit(0))

pcm_names = subprocess.run(["aplay", "-L"], stdout=subprocess.PIPE).stdout.decode(
    "utf-8"
)

print("ALSA PCM Names:\n")
print(pcm_names)

pcm = input(
    "Please enter the Name of the hw PCM you wish to use (for example hw:CARD=D10s,DEV=0): "
)

if not pcm.startswith("hw:"):
    print("Error: Invalid hw PCM: {}".format(pcm))
    sys.exit(1)

try:
    card = pcm.replace("hw:CARD=", "").strip()
    [card, dev] = card.split(",")
    dev = dev.replace("DEV=", "").strip()

except:
    print("Error: Invalid hw PCM: {}".format(pcm))
    sys.exit(1)

if not card or not dev:
    print("Error: Invalid hw PCM: {}".format(pcm))
    sys.exit(1)

try:
    dev = int(dev)

except:
    print("Error: Invalid hw PCM: {}".format(pcm))
    sys.exit(1)

result = subprocess.run(
    [
        "aplay",
        "-D{}".format(pcm),
        "--dump-hw-params",
        "/usr/share/sounds/alsa/Front_Right.wav",
    ],
    stderr=subprocess.PIPE,
    stdout=subprocess.PIPE,
)

out = result.stdout.decode("utf-8")
err = result.stderr.decode("utf-8")
hw_params = out or err

params_list = []

for line in hw_params.split("\n"):
    if line.startswith("FORMAT:") or line.startswith("RATE:"):
        params_list.append(line)

if params_list:
    print("\n{} Supported Rate(s) and Format(s):\n".format(pcm))
    print("\n".join(params_list))

else:
    print(hw_params)
    sys.exit(1)

format_ = input("Please enter the desired supported format: ")

if not format_.startswith("S"):
    print("Error: Invalid format: {}".format(format_))
    sys.exit(1)

rate = input("Please enter the desired supported sampling rate: ")

try:
    rate = int(rate)

except:
    print("Error: Invalid sampling rate: {}".format(rate))
    sys.exit(1)

file_data = """# /etc/asound.conf

pcm.!default {{
    type plug
    slave.pcm {{
        type dmix
        ipc_key {{
            @func refer
            name defaults.pcm.ipc_key
        }}
        ipc_gid {{
            @func refer
            name defaults.pcm.ipc_gid
        }}
        ipc_perm {{
            @func refer
            name defaults.pcm.ipc_perm
        }}
        slave {{
            pcm {{
                type hw
                card {0}
                device {1}
                nonblock {{
                    @func refer
                    name defaults.pcm.nonblock
                }}
            }}
            channels 2
            rate {2}
            format {3}
        }}
        bindings {{
            0 0
            1 1
        }}
    }}
}}

ctl.!default {{
    type hw
    card {0}
}}""".format(
    card, dev, rate, format_
)

file_path = "/etc/asound.conf"

if os.path.exists(file_path):
    print(
        "Error: /etc/asound.conf already exists please backup and rename it or delete it."
    )
    sys.exit(1)

try:
    with open(file_path, "w") as f:
        f.write(file_data)

except Exception as e:
    print("Error: {} Hint: try calling this script with sudo".format(e))
    sys.exit(1)

else:
    print("/etc/asound.conf written successfully please verify that it is correct")

```