dtcooper/raspotify #572: Raspotify won't work with Bluetooth Device

github-gabriel, 2022-09-18:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hello,
I have setup Raspotify on my Raspberry Pi 3B+. I can also connect to it and play Spotify, but I only hear the music when I use my headphones, which are connected over the headphone jack. If I try my bluetooth headphones I can't hear anything but on my mobile device it shows that it's playing. Raspotify seems fine when checking the status
```pi@raspberrypi:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-09-18 15:50:48 CEST; 1min 33s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
   Main PID: 2404 (librespot)
      Tasks: 2 (limit: 1596)
        CPU: 7.310s
     CGroup: /system.slice/raspotify.service
             └─2404 /usr/bin/librespot

Sep 18 15:50:48 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Sep 18 15:50:48 raspberrypi librespot[2404]: [2022-09-18T13:50:48Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Sep 18 15:50:48 raspberrypi librespot[2404]: [2022-09-18T13:50:48Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Sep 18 15:50:48 raspberrypi librespot[2404]: [2022-09-18T13:50:48Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Sep 18 15:51:08 raspberrypi librespot[2404]: [2022-09-18T13:51:08Z WARN  librespot_playback::player] Player::play called from invalid state
Sep 18 15:51:18 raspberrypi librespot[2404]: Failed to get home directory.
```
(Here are some images from VNC Viewer of the different devices)
If I connect my bluetooth device like this, listening to music over Raspotify won't work (But if I try my bluetooth device over something like YouTube on the Raspberry I can hear the audio from the YouTube video)
[image]
But if I set the audio output to the headphone jack and use my other headphones that have a headphone jack (not the same device as the bluetooth device!) like this, I can hear music from Raspotify
[image]
So for me, Raspotify "works over the headphone jack but not with the bluetooth device". If you need additional information just tell me ^^ 
(And sorry for my English, I am not Native English)

JasonLG1979, 2022-09-18:
That's a known behavior. The [README](https://github.com/dtcooper/raspotify#readme) clearly states that Raspotify is not intended for desktop use.

Raspotify is a system level daemon it neither knows nor cares about your desktop audio settings or preferences. It simply will try to use the system's default audio output/device or the one you tell it to in the config. I'm honestly surprised the service isn't crashing more often?

With a little research, a text editor and the librespot binary provided by the package you can create your own user level daemon that will honor your user audio preferences (it has the pulseaudio backend compiled into it to allow for that). But that's well beyond the scope of this project and completely unsupported.

closed 2022-09-18 by github-actions[bot]

JasonLG1979, 2022-09-18:
For reference:

https://github.com/librespot-org/librespot/blob/dev/contrib/librespot.user.service

https://github.com/librespot-org/librespot/wiki/Options

https://www.freedesktop.org/software/systemd/man/systemd.unit.html