dtcooper/raspotify #573: Some problem with update raspotify on raspberry

szygmunt, 2022-09-19:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi,

i have problem with upgrade raspotify to latest version through apt on raspios:

https://dtcooper.github.io/raspotify raspotify/main armhf raspotify armhf 0.42.5~librespot.v0.4.2-6537c44
  404  Not Found [IP: 185.199.109.153 443]
Is it some problem on your site or I have to check something on my own?

JasonLG1979, 2022-09-19:
0.42.6 is the latest version

JasonLG1979, 2022-09-19:
I don't know what to tell you? I removed raspotify, cleared my apt cache and reinstalled it from the repo just fine. Maybe it was a github hiccup?

JasonLG1979, 2022-09-20:
I'm closing this. There's nothing I can do about the server availability of the repo.

closed 2022-09-20 by JasonLG1979