dtcooper/raspotify #583: Can't get ALSA EQ Plugn work on RPi3 B+

blakyris, 2022-10-30:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi everyone,

I installed Ubuntu Server on my RPi3 B+ with Raspotify.
Everything works execpt the ALSA EQ plugin.
I think the problem is the user used by the systemd service.

When I start librespot as sudo without any options it defaults on the EQ.
But when I use raspotify the EQ is not applied.

Here is my asound.conf

```
pcm.!default {
    type plug
    slave.pcm plugequal
}

ctl.!default {
    type hw card 0
}

ctl.equal {
    type equal
    controls /tmp/.alsaequal.bin
}

pcm.plugequal {
    type equal
    controls /tmp/.alsaequal.bin
    slave.pcm "plughw:0,0"
}

pcm.equal {
    type plug
    slave.pcm plugequal
}
```

In the config file :
I tried with `OPTIONS=--device "equal"` and without but nothing happened.
I set DEVICE to `equal` and `default` nothing changes also.

This problem appear to be the same with Shairport on my config. I don't know what I'm doing wrong.

Thanks for your help.

JasonLG1979, 2022-10-30:
Duplicate of https://github.com/dtcooper/raspotify/issues/23 and  https://github.com/dtcooper/raspotify/issues/475 and https://github.com/dtcooper/raspotify/issues/569

closed 2022-10-30 by github-actions[bot]

JasonLG1979, 2022-10-30:
In the future please search though the issues to avoid filing a duplicate bug report.