dtcooper/raspotify #584: Disconnect after play on bullseye

kugel89, 2022-10-31:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Raspotify stopped working on my Pi4 with raspbian 10 - so I upgraded to raspbian 11. I followed the guide at https://github.com/dtcooper/raspotify/wiki/Raspotify-with-PulseAudio-as-a-system-service. However, I cannot get raspotify to play any music. After I connect and start playback the connection drops.

These are the messages I find when running journalctl --system _SYSTEMD_UNIT=raspotify.service -b
 
Oct 31 19:22:21 kugels-pi librespot[541]: [2022-10-31T18:22:21Z WARN  librespot] Could not initialise discovery: Setting up dns-sd failed: No such device (os error 19).
Oct 31 19:22:21 kugels-pi librespot[541]: [2022-10-31T18:22:21Z ERROR librespot] Discovery is unavailable and no credentials provided. Authentication is not possible.
Oct 31 19:57:39 kugels-pi librespot[1072]: [2022-10-31T18:57:39Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Oct 31 19:57:39 kugels-pi librespot[1072]: [2022-10-31T18:57:39Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Oct 31 19:57:39 kugels-pi librespot[1072]: [2022-10-31T18:57:39Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Oct 31 19:57:39 kugels-pi librespot[1072]: [2022-10-31T18:57:39Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gue1.spotify.com:4070
Oct 31 19:57:46 kugels-pi librespot[1072]: Failed to get home directory.
Oct 31 19:57:46 kugels-pi librespot[1072]: Failed to get home directory.
Oct 31 19:57:46 kugels-pi librespot[1072]: Failed to load cookie file from cookie: No such file or directory
Oct 31 19:57:46 kugels-pi librespot[1072]: Failed to get home directory.
Oct 31 19:57:46 kugels-pi librespot[1072]: Failed to get home directory.
Oct 31 19:57:46 kugels-pi librespot[1072]: Failed to load cookie file from cookie: No such file or directory
Oct 31 19:57:46 kugels-pi librespot[1072]: ALSA lib pulse.c:242:(pulse_connect) PulseAudio: Unable to connect: Access denied
Oct 31 19:57:46 kugels-pi librespot[1072]: [2022-10-31T18:57:46Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or A

I do not understand why the cookie file is mentioned as I have disabled it in the referenced setup guide for pulseaudio - or maybe it is another cookie?

After trying to fix it for hours I hope that somebody here can help - thanks!

JasonLG1979, 2022-10-31:
Are you using the Pi as a Desktop or in headless mode?

kugel89, 2022-10-31:
I am using it with a desktop environment even though I rarely use it.

JasonLG1979, 2022-10-31:
I'm by no means a PulseAudio expert but one thing that does stick out is:
```
Oct 31 19:57:46 kugels-pi librespot[1072]: Failed to get home directory.
```

If PulseAudio isn't going to find a Raspotify `~/` (home directory) because it doesn't exist. The Raspotify service runs as a dynamic user, dynamic users don't have homes.

> I am using it with a desktop environment even though I rarely use it.

If you rarely use the Desktop then I would suggest just installing the headless version of  Raspbian 11 and using the ALSA backend (the default) with Raspotify. It's a much smoother experience. The PulseAudio backend is included for advanced users, batteries not included so to speak.

JasonLG1979, 2022-10-31:
I meant to say *If you rarely don't use the Desktop...*

kugel89, 2022-10-31:
Maybe that would be the best solution. I wanted to use it initially but found it too unresponsive. I'll consider doing that

JasonLG1979, 2022-10-31:
> Maybe that would be the best solution. I wanted to use it initially but found it too unresponsive. I'll consider doing that

Yep, for as much as they say a Pi 4 can function as a desktop in my experience not so much. I never run the desktop version except to check it out when new Pi's come out. Other than that I just run them headless and ssh into them.

closed 2022-10-31 by JasonLG1979