dtcooper/raspotify #585: Raspotify crash and full disc

szygmunt, 2022-11-03:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

I have problem with raspotify running on Rpi 3B+. Raspotify crash as you can see in my log and countless write to syslog and daemon.log in /var/log directory. After few hours whole sd card is full. I have this problem since few weeks.

### Relevant log output

```shell
Nov  1 11:49:23 pi librespot[2597]: [2022-11-01T10:49:23Z WARN  librespot_playback::player] <Ten o Polsce> is not available in any supported format
Nov  1 11:49:23 pi librespot[2597]: [2022-11-01T10:49:23Z WARN  librespot_playback::player] <Ten o Polsce> is not available in any supported format
Nov  1 11:49:23 pi librespot[2597]: [2022-11-01T10:49:23Z WARN  librespot_playback::player] <Ten o Polsce> is not available in any supported format
Nov  1 11:49:23 pi librespot[2597]: [2022-11-01T10:49:23Z WARN  librespot_playback::player] <Ten o Polsce> is not available in any supported format
Nov  1 11:49:23 pi librespot[2597]: [2022-11-01T10:49:23Z WARN  librespot_playback::player] <Ten o Polsce> is not available in any supported format
Nov  1 11:49:24 pi librespot[2597]: [2022-11-01T10:49:24Z WARN  librespot_playback::player] <Ten o Polsce> is not available in any supported format
Nov  1 11:49:24 pi librespot[2597]: [2022-11-01T10:49:24Z WARN  librespot_playback::player] Skipping to next track, unable to load track <SpotifyId { id: 245544158687370
939398471321314401245500, audio_type: Podcast }>: ()
Nov  1 11:49:24 pi librespot[2597]: thread '<unnamed>' panicked at 'Map must not be polled after it returned `Poll::Ready`', /build/cache/registry/src/github.com-1ecc629
9db9ec823/futures-util-0.3.21/src/future/future/map.rs:62:17
Nov  1 11:49:24 pi librespot[2597]: stack backtrace:
Nov  1 11:49:24 pi librespot[2597]:    0:   0x795fbc - std::backtrace_rs::backtrace::libunwind::trace::h3ecd63e968c83846
Nov  1 11:49:24 pi librespot[2597]:                        at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/../../backtrace/src/backtrace/libunwind.rs:
93:5
Nov  1 11:49:24 pi librespot[2597]:    1:   0x795fbc - std::backtrace_rs::backtrace::trace_unsynchronized::h896664514e76113d
Nov  1 11:49:24 pi librespot[2597]:                        at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
Nov  1 11:49:24 pi librespot[2597]:    2:   0x795fbc - std::sys_common::backtrace::_print_fmt::h32a493fae6d6e47d
Nov  1 11:49:24 pi librespot[2597]:                        at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/sys_common/backtrace.rs:66:5
Nov  1 11:49:24 pi librespot[2597]:    3:   0x795fbc - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h89340f3152c95102
Nov  1 11:49:24 pi librespot[2597]:                        at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/sys_common/backtrace.rs:45:22
Nov  1 11:49:24 pi librespot[2597]:    4:   0x795fbc - std::backtrace_rs::backtrace::libunwind::trace::h3ecd63e968c83846
Nov  1 11:49:24 pi librespot[2597]:                        at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/../../backtrace/src/backtrace/libunwind.rs:
93:5
Nov  1 11:49:24 pi librespot[2597]:    5:   0x795fbc - std::backtrace_rs::backtrace::trace_unsynchronized::h896664514e76113d
Nov  1 11:49:24 pi librespot[2597]:                        at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
Nov  1 11:49:24 pi librespot[2597]:    6:   0x795fbc - std::sys_common::backtrace::_print_fmt::h32a493fae6d6e47d
Nov  1 11:49:24 pi librespot[2597]:                        at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/sys_common/backtrace.rs:66:5
Nov  1 11:49:24 pi librespot[2597]:    7:   0x795fbc - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h89340f3152c95102
Nov  1 11:49:24 pi librespot[2597]:                        at /rustc/a55dd71d5fb0ec5a6a3a9e8c27b2127ba491ce52/library/std/src/sys_common/backtrace.rs:45:22
```

JasonLG1979, 2022-11-03:
Podcast support is sketchy at best in librespot. Clear your logs and stop trying to listen to that particular podcast with Raspotify. Librespot only supports ogg vorbis streams. If I had to guess that podcast is probably MP3.

JasonLG1979, 2022-11-03:
I just pushed a new release that although still won't allow you to play the offending podcast it hopefully shouldn't panic anymore.

JasonLG1979, 2022-11-15:
I'm going to close this because of a lack of a response.

closed 2022-11-15 by JasonLG1979