dtcooper/raspotify #590: Failed installation because dependencies are not on the required versions

reinisb, 2022-11-24:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi,

I'm on RPi4 running the latest 32bit Raspberry Pi OS (Lite) version. When I run Raspotify installation command it fails because the required packages are not at the versions required by Raspotify. When I then try to update each individual dependency it says that I already have the latest version (but Raspotify requires a newer version).

Is this by design that I need to manually find and install the latest version of each dependency (even if my system tells me that I already have the latest version)? Or is this a side effect of me using 32bit version of Raspberry PI OS? Thanks!

> pi@wireguard:~ $ sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> curl is already the newest version (7.64.0-4+deb10u3).
> 0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
> 
> Unmet minimum required package version(s):
> 
> libc6 >= 2.31 is required but 2.28-10+rpt2+rpi1+deb10u1 is installed.
> systemd >= 247.3 is required but 241-7~deb10u8+rpi1 is installed.
> libasound2 >= 1.2.4 is required but 1.1.8-1+rpt2 is installed.
> alsa-utils >= 1.2.4 is required but 1.1.8-2+rpt1 is installed.
> libpulse0 >= 14.2 is required but 12.2-4+deb10u1+rpt3 is installed.
> 
> Please make sure you are running a compatible armhf (ARMv7), arm64, or amd64 Debian based OS.

> pi@wireguard:~ $ sudo apt-get install libasound2
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> libasound2 is already the newest version (1.1.8-1+rpt2).

JasonLG1979, 2022-11-24:
Duplicate of https://github.com/dtcooper/raspotify/issues/565, https://github.com/dtcooper/raspotify/issues/564, https://github.com/dtcooper/raspotify/issues/511, and 588.

closed 2022-11-24 by JasonLG1979

reinisb, 2022-11-24:
@JasonLG1979 Gotcha! It's a shame Raspberry Pi OS is not supported any longer (even though its based on debian 11). I've been using Raspotify on my Raspberry Pi OS for years, but will not be able to any longer. Really sad :(

reinisb, 2022-11-24:
Nevermind, mine is still on Buster. Will try to find a way to upgrade it to Bullseye.