dtcooper/raspotify #591: How do I get my $PLAYER_EVENT to work?

mommotti, 2022-11-29:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

**I saved this on /etc/default/onscript.sh and did the chmod**
![image](https://user-images.githubusercontent.com/49097348/204574038-3ea1353c-03d7-4c54-b9f7-3d3b3e2aed92.png

**off.py and on.py are on /home/pi/scripts/ and both scripts work (one script is to turn on 3 blinking leds and the other is to turn it off)**

**This is the conf file:**
[image]

I may be failed with the conf file syntax.

Thank you in advance :)

JasonLG1979, 2022-11-29:
That will not work for a couple reasons both related to the fact that the Raspotify service runs as an unprivileged dynamic user. Because of that it can't access that location and sudo will not work.

The troubleshooting shooting guide in the wiki (the one you said you read when you checked the box to file this bug report) says as much.

closed 2022-11-29 by JasonLG1979

JasonLG1979, 2022-11-29:
You're args are also incorrect.

It should be:
```
LIBRESPOT_EMIT_SINK_EVENTS=
LIBRESPOT_ONEVENT="/path/to/your/script"
```

JasonLG1979, 2022-11-29:
You can use triple backticks to quite terminal output.

```
like this
```

And you can use triple backticks followed by the language name for syntax highlighting:

```python
def foo():
    return bar
```