dtcooper/raspotify #595: Start of service fails because /usr/bin/sleep does not exist

kaihs, 2022-12-04:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

After installing the latest version start the raspotify service fails with an error because /usr/bin/sleep does not exist.

This is on a debian based Armbian system 
$ cat /etc/debian_version 
11.5

$ which sleep
/bin/sleep

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any

```shell
-- Journal begins at Fri 2022-12-02 20:18:30 CET, ends at Sun 2022-12-04 12:19:47 CET. --
Dez 04 11:59:43 bananapi systemd[1]: Stopping Raspotify (Spotify Connect Client)...
Dez 04 11:59:43 bananapi systemd[1]: raspotify.service: Succeeded.
Dez 04 11:59:43 bananapi systemd[1]: Stopped Raspotify (Spotify Connect Client).
Dez 04 11:59:43 bananapi systemd[1]: raspotify.service: Consumed 37.244s CPU time.
Dez 04 11:59:57 bananapi systemd[1]: Starting Raspotify (Spotify Connect Client)...
Dez 04 11:59:57 bananapi systemd[20514]: raspotify.service: Failed to locate executable /usr/bin/sleep: No such file or directory
Dez 04 11:59:57 bananapi systemd[20514]: raspotify.service: Failed at step EXEC spawning /usr/bin/sleep: No such file or directory
Dez 04 11:59:57 bananapi systemd[1]: raspotify.service: Control process exited, code=exited, status=203/EXEC
Dez 04 11:59:57 bananapi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Dez 04 11:59:57 bananapi systemd[1]: Failed to start Raspotify (Spotify Connect Client).
Dez 04 11:59:57 bananapi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
Dez 04 11:59:59 bananapi systemd[1]: Starting Raspotify (Spotify Connect Client)...
Dez 04 11:59:59 bananapi systemd[20623]: raspotify.service: Failed to locate executable /usr/bin/sleep: No such file or directory
Dez 04 11:59:59 bananapi systemd[20623]: raspotify.service: Failed at step EXEC spawning /usr/bin/sleep: No such file or directory
Dez 04 11:59:59 bananapi systemd[1]: raspotify.service: Control process exited, code=exited, status=203/EXEC
Dez 04 11:59:59 bananapi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Dez 04 11:59:59 bananapi systemd[1]: Failed to start Raspotify (Spotify Connect Client).
Dez 04 11:59:59 bananapi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
```

JasonLG1979, 2022-12-04:
On both Debian and Raspberry Pi OS `which sleep` returns `/usr/bin/sleep`, but `/bin/sleep` also works because `/bin` is link to `/usr/bin` on both and `sleep` actually is in `/bin`.

I will change it to `/bin/sleep` but if I were you I'd ask Armbian why `/bin` and `/usr/bin` aren't linked. That in itself means that Armbian isn't 100% Debian compatible.

closed 2022-12-04 by JasonLG1979

JasonLG1979, 2022-12-04:
0.43.13 has been released with a fix.

kaihs, 2022-12-06:
My system has been updated for years and has never been reinstalled from scratch.
It might be that through all those updates the usr merge was never executed.

To avoid more possible future problems caused by this I should probably perform the merge manually by installing package usrmerge.