dtcooper/raspotify #597: Can`t find Raspotify on device list

jackobtone, 2022-12-05:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

I will hook up to this topic. Recently I`ve installed DietPi + Alsa + Raspotify. Everything worked till todays update (got a prompt in the system). After it i cant find Raspberry in Spotify app. I have reinstalled the system, installed Alsa, installed Raspotify just like I did before and it didnt work. Command _sudo service raspotify status_ give back a message:

>root@DietPi:~# sudo service raspotify status
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; disabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-12-05 18:39:34 GMT; 46s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 462 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
   Main PID: 522 (librespot)
      Tasks: 1 (limit: 4531)
        CPU: 161ms
     CGroup: /system.slice/raspotify.service
             └─522 /usr/bin/librespot
Dec 05 18:39:24 DietPi systemd[1]: Starting Raspotify (Spotify Connect Client)...
Dec 05 18:39:34 DietPi systemd[1]: Started Raspotify (Spotify Connect Client).

I assume everything should work, but it doesnt

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any

```shell
root@DietPi:~# sudo journalctl -u raspotify -b
-- Journal begins at Mon 2022-12-05 18:39:12 GMT, ends at Mon 2022-12-05 18:40:55 GMT. --
Dec 05 18:39:24 DietPi systemd[1]: Starting Raspotify (Spotify Connect Client)...
Dec 05 18:39:34 DietPi systemd[1]: Started Raspotify (Spotify Connect Client).
```

JasonLG1979, 2022-12-05:
Is the device you are trying to connect from on the same local network?

Can you ssh into the device?

What is the output of `librespot -v` I'm mostly concerned with things to do with zeroconf/mdns discovery.

jackobtone, 2022-12-05:
1. Yes, it is in the same local network. I have Asus AiMesh setup, Raspberry CM4 is connected directly to one of the routers. 
2. No idea what You mean, I`m green
3. 

> root@DietPi:~# librespot -v
[2022-12-05T18:52:20Z INFO  librespot] librespot 0.4.2 86743ea (Built on 2022-12-04, Build ID: 8yRuTOl9, Profile: release)
[2022-12-05T18:52:20Z TRACE librespot] Command line argument(s):
[2022-12-05T18:52:20Z TRACE librespot]          v
[2022-12-05T18:52:20Z DEBUG librespot_discovery::server] Zeroconf server listening on 0.0.0.0:38119
^C[2022-12-05T18:52:26Z INFO  librespot] Gracefully shutting down

4. Currently settings in terms of conf file are stock, didnt change anything in it after problems occured.

JasonLG1979, 2022-12-05:
ssh stands for secure shell. You can use ssh to get a remote terminal so you can admin the device from another computer.

Is it possible that your router is blocking mdns?

jackobtone, 2022-12-05:
Well, all the logs Ive pasted here are from Putty, so yes, I can ssh

Well, no idea. Main router is Asus AX89X, the seondary with which Raspberry is connected is AX86U. How can I check if they are blocking mdns?

JasonLG1979, 2022-12-05:
The default behavior of the discovery code hasn't really changed in a long time but I did very recently backport a feature that allows a user to specify an IP address that mdns binds to. But that hasn't been an issue for me at all and the default behavior when that option is not set is exactly the same as before.

JasonLG1979, 2022-12-05:
To clarify zeroconf was working fine for you before and now it's not?

JasonLG1979, 2022-12-05:
Is there a firewall enabled on the device that's running librespot?

jackobtone, 2022-12-05:
When it comes to zeroconf - earlier everything worked properly, without any issues, so I guess it worked, I didnt check it before problems occured, because there wasnt such need.

No, no firewall on DietPi/Raspberry CM4, but ofcourse firewall is set on on the router

JasonLG1979, 2022-12-05:
If you haven't changed the firewall on the router and it worked before it should still work I imagine? You didn't have any specific rules before for the device before you reinstalled?

jackobtone, 2022-12-05:
No, progress of reinstallation looked exactly the same as at the first time. 
flashing software onto CM4 eMMC, startup, installing Alsa and Raspotify via DietPi (it has inbuilt list of software and installs all pointed software by itself). Install suceeded, no errors.

JasonLG1979, 2022-12-05:
I mean on your router. Were/are there any device or protocol specific rules in place?

JasonLG1979, 2022-12-05:
The next suggestion seems kinda silly but maybe power cycle everything.

jackobtone, 2022-12-05:
> I mean on your router. Were/are there any device or protocol specific rules in place?

No, nothing like that.

> This next suggestion seems kinda silly but maybe power cycle everything.

Nothing to loose, will give it a go :)

JasonLG1979, 2022-12-05:
The next steps are for me to install dietpi on my Pi and see if I can recreate the issue and for you to manually install an older release prior to the backport to see if that's the problem. If it is the problem I'm pretty sure I can approximate the feature with some systemd magic, revert the backport and issue a new releas.

jackobtone, 2022-12-05:
Meanwhile I will reinstall DietPi once again. During first system boot I used to do it with my Raspberry connected to the other router than the target one. Maybe this causes some trouble? But on the other hand I did exactly the same when was setting it up for the very first time and everything worked. Anyway i will give it a go

jackobtone, 2022-12-05:
Fresh install - the additional update packages that DietPi installed itself during first startup

> Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils binutils-aarch64-linux-gnu binutils-common libbinutils
  libctf-nobfd0 libctf0 libexpat1 libmpdec3 libpci3 libpython3-stdlib
  libpython3.9-minimal libpython3.9-stdlib media-types pci.ids pciutils
  python3 python3-minimal python3.9 python3.9-minimal
Suggested packages:
  binutils-doc python3-doc python3-tk python3-venv python3.9-venv
  python3.9-doc binfmt-support
Recommended packages:
  flashrom
The following NEW packages will be installed:
  binutils binutils-aarch64-linux-gnu binutils-common libbinutils
  libctf-nobfd0 libctf0 libexpat1 libmpdec3 libpci3 libpython3-stdlib
  libpython3.9-minimal libpython3.9-stdlib media-types pci.ids pciutils
  python3 python3-minimal python3.9 python3.9-minimal rpi-eeprom

Decided, that I will install Alsamixer and Raspotify separetelly, begun with Alsa. After Alsa installation 

> root@DietPi:~# alsamixer
cannot open mixer: No such file or directory
root@DietPi:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 1: Audio [McIntosh HD USB Audio], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

Installed Raspotify:

>  DietPi-Software
─────────────────────────────────────────────────────
 Step: Installing Raspotify: Spotify connect client
[  OK  ] DietPi-Software | Checking URL: https://dtcooper.github.io/raspotify/key.asc
[  OK  ] DietPi-Software | eval curl -sSfL 'https://dtcooper.github.io/raspotify/key.asc' | gpg --dearmor -o /etc/apt/trusted.gpg.d/dietpi-raspotify.gpg --yes
[  OK  ] DietPi-Software | eval echo 'deb https://dtcooper.github.io/raspotify/ raspotify main' > /etc/apt/sources.list.d/raspotify.list
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://deb.debian.org/debian bullseye InRelease
Hit:2 https://deb.debian.org/debian bullseye-updates InRelease
Hit:3 https://deb.debian.org/debian-security bullseye-security InRelease
Hit:4 https://deb.debian.org/debian bullseye-backports InRelease
Hit:5 https://archive.raspberrypi.org/debian bullseye InRelease
Get:6 https://dtcooper.github.io/raspotify raspotify InRelease [3143 B]
Get:7 https://dtcooper.github.io/raspotify raspotify/main arm64 Packages [915 B]
Fetched 4058 B in 2s (1641 B/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update
[ INFO ] DietPi-Software | APT install for: raspotify, please wait...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libasyncns0 libflac8 libogg0 libopus0 libpulse0 libsndfile1 libvorbis0a
  libvorbisenc2 libwrap0 libxau6 libxcb1 libxdmcp6
Suggested packages:
  opus-tools pulseaudio
The following NEW packages will be installed:
  libasyncns0 libflac8 libogg0 libopus0 libpulse0 libsndfile1 libvorbis0a
  libvorbisenc2 libwrap0 libxau6 libxcb1 libxdmcp6 raspotify
0 upgraded, 13 newly installed, 0 to remove and 0 not upgraded.
Need to get 2813 kB of archives.
After this operation, 3565 kB of additional disk space will be used.
Get:1 https://deb.debian.org/debian bullseye/main arm64 libasyncns0 arm64 0.8-6+b2 [12.5 kB]
Get:2 https://deb.debian.org/debian bullseye/main arm64 libogg0 arm64 1.3.4-0.1 [26.3 kB]
Get:3 https://deb.debian.org/debian bullseye/main arm64 libflac8 arm64 1.3.3-2+deb11u1 [95.1 kB]
Get:4 https://deb.debian.org/debian bullseye/main arm64 libopus0 arm64 1.3.1-0.1 [169 kB]
Get:5 https://deb.debian.org/debian bullseye/main arm64 libvorbis0a arm64 1.3.7-1 [86.2 kB]
[... 81 lines cut ...]
[  OK  ] DietPi-Services | restart : raspotify
[  OK  ] DietPi-Services | restart : cron

Raspotify test:

> root@DietPi:~# sudo service raspotify status
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; disabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-12-05 22:05:38 GMT; 5min ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 5275 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
   Main PID: 5276 (librespot)
      Tasks: 1 (limit: 4531)
        CPU: 229ms
     CGroup: /system.slice/raspotify.service
             └─5276 /usr/bin/librespot
Dec 05 22:05:28 DietPi systemd[1]: Starting Raspotify (Spotify Connect Client)...
Dec 05 22:05:38 DietPi systemd[1]: Started Raspotify (Spotify Connect Client).

Alsamixer works only under: "alsamixer -c 1", doesnt work under just "alsamixer", but it did before.

I believe it`s not Raspotify fault, but rather something with DietPi OS

JasonLG1979, 2022-12-05:
> Alsamixer works only under: "alsamixer -c 1", doesnt work under just "alsamixer", but it did before.
> 
> I believe it`s not Raspotify fault, but rather something with DietPi OS

Yep, Raspotify doesn't have anything to do with that.

Maybe just try Raspberry Pi OS Lite (64 bit) ?

JasonLG1979, 2022-12-05:
> USB DAC in system settings, "alsamixer" works, but still Raspotify isnt on device list

Raspotify wouldn't be on the device list. It's not a device.

JasonLG1979, 2022-12-05:
At this point I'm not seeing a reason to have any faith in DietPi.

jackobtone, 2022-12-05:
> > USB DAC in system settings, "alsamixer" works, but still Raspotify isnt on device list
> 
> Raspotify wouldn't be on the device list. It's not a device.

Yeah yeah, sorry, it was a mental shortcut. I mean my Raspberry based file transport isnt detected by Spotify Connect

jackobtone, 2022-12-05:
I will try Raspberry Pi OS as you sugested

JasonLG1979, 2022-12-05:
When you get that installed I'll walk you though configuring everything.

JasonLG1979, 2022-12-05:
Make sure to install the "Lite" version so it's headless.

JasonLG1979, 2022-12-05:
Here's a link to the official iso imager https://downloads.raspberrypi.org/imager/imager_latest.exe

You can click the little "gear" icon to preconfigure things like wifi, hostname, username password, enable ssh and whatnot.

jackobtone, 2022-12-05:
Writing in progress. After it is complete, to install alsa i should use this code?

> sudo apt update
sudo apt install alsa-utils libasound2-plugins

JasonLG1979, 2022-12-05:
> Writing in progress. After it is complete, to install alsa i should use this code?
> 
> > sudo apt update
> > sudo apt install alsa-utils libasound2-plugins

You don't need to do that. Raspberry Pi OS already includes most if not all you need from ALSA and the Raspotify package will pull in whatever else it needs.

JasonLG1979, 2022-12-05:
You only need `libasound2-plugins` if you your DAC doesn't support a sampling rate of 44100. If your DAC does there is absolutely no point in resampling. Resampling NEVER improves sound quality.

If your DAC doesn't support 44100 then I would install `libasound2-plugins` with:
```
sudo apt install -y --no-install-recommends libasound2-plugins
```
`--no-install-recommend` prevents shit you don't need from being installed.

jackobtone, 2022-12-05:
That`s awesome to know. Never used resampling. DAC accepts all formats up to DSD256

JasonLG1979, 2022-12-05:
If librespot is the only thing that's going to use the DAC we can set up librespot to talk directly to the DAC.

JasonLG1979, 2022-12-05:
librespot supports `S16`, `S24`, `S24_3`, `S32`, `F32`, and `F64` at 44100. For bit depth you want the highest your DAC will support.

The reason being is that Spotify streams lossy audio and lossy audio doesn't have a real bit depth it has to be quantized back to having a bit depth while converting it back to PCM so the higher the bit depth the less quantization noise and the more headroom for digital volume control.

jackobtone, 2022-12-05:
System woke up. So what first should I do?
I would also like to use my file transport as Roon Bridge

jackobtone, 2022-12-05:
Im curious if it will work, because some of the systems i tried didnt work with PCIe JCat Femto USB card connected to Raspberry. Anyway system woke up, thats a good sign

JasonLG1979, 2022-12-05:
> I would also like to use my file transport as Roon Bridge

Then we will need to set up dmix. You'll want to install `libasound2-plugins` as mentioned before that way you can accept different sampling rates.

JasonLG1979, 2022-12-05:
So go ahead and do a `sudo apt update && sudo apt dist-upgrade -y`

jackobtone, 2022-12-05:
Its done

JasonLG1979, 2022-12-05:
I just looked up the PCIe JCat Femto USB card. That thing is pure snake oil. If it doesn't work just unplug it and just use one of the built in usb ports for your DAC.

JasonLG1979, 2022-12-05:
Now install Raspotify:

```
sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
```

jackobtone, 2022-12-05:
Its done

JasonLG1979, 2022-12-05:
```
sudo apt install -y --no-install-recommends libasound2-plugins
```

If you haven't already.

JasonLG1979, 2022-12-05:
Does Raspotify show up on a controlling device on your network?

jackobtone, 2022-12-05:
Unfortunatelly it doesn`t. I will restart router and raspberry

JasonLG1979, 2022-12-05:
Well shit.

JasonLG1979, 2022-12-05:
Let's rule out that backport.

Remove the current version:
```
sudo apt purge -y raspotify
```

***Assuming you installed the 64 bit version*** 

Download the version before the backport:

```
curl -O -sL https://github.com/dtcooper/raspotify/releases/download/0.43.9/raspotify_0.43.9.librespot.v0.4.2-eb5e1a3_arm64.deb
```

Install that version:
```
sudo apt install -y ./raspotify_0.43.9.librespot.v0.4.2-eb5e1a3_arm64.deb
```

JasonLG1979, 2022-12-05:
After that's done check to see if Raspotify shows up, and we can go from there.

jackobtone, 2022-12-05:
Nope, still nothing

JasonLG1979, 2022-12-05:
Well that's good news for me (I don't have to revert it,lol) and sorta bad news for you.

Now purge the old version and reinstall the latest version again:

```
sudo apt purge -y raspotify && sudo apt install -y raspotify
```

After that we're going to just disable discovery and use your creds. The downside to that is that it basically makes it a private player. No one else but that account will be able to see it.

jackobtone, 2022-12-05:
I will reset this router

JasonLG1979, 2022-12-05:
A neat side effect though is you don't have to be on the same network anymore to control it.

JasonLG1979, 2022-12-05:
Go to your account overview page.

If you're in the US:

https://www.spotify.com/us/account/overview/

Or you can click the the account link in the web client:
[image]

jackobtone, 2022-12-05:
I know my login and password, and i think i know what else I should do. I`m installing the newest version of Raspotify now. Then will edit conf file. If something fails I will let you know for sure :) 

Thanks a lot for all your help, you are doing really good job with this project. Just keep on!

JasonLG1979, 2022-12-05:
It's not that same as your login info.

Scroll down and look for your "username"

[image]

And then go to the device password page:

[image]

Click this:
[image]

Set your device password:
[image]

jackobtone, 2022-12-06:
Ok, Raspotify is visible again and works as it should. But what actually causes such problems?

JasonLG1979, 2022-12-06:
After you put in your creds or just out of the blue?

JasonLG1979, 2022-12-06:
Ok, Raspotify is visible again and works as it should. But what actually causes such problems?

Shit mdns implementations on actually a lot of wifi routers. My TP Link wifi router had similar problems. I ended up just buying a standalone router and just using the TP Link as a wifi AP.

JasonLG1979, 2022-12-06:
It's also up in the air if libmdns (what librespot uses for mdns) is completely spec compliant?

jackobtone, 2022-12-06:
Wow, I just tried another thing and it worked. I have uploaded onto Raspberry a "stock" conf file without Spotify password and login, logged into my router and fixed/connected MAC adress with IP and also changed previous IP adress of my Raspberry. It worked and Pi showed up on the device list, without necessity of logging in

JasonLG1979, 2022-12-06:
That's because it's reading the previously cached creds.

jackobtone, 2022-12-06:
Even after Raspbian reboot?

JasonLG1979, 2022-12-06:
Yes. The token is saved to disk. That will work until the token expires and you'll be back to square one.

jackobtone, 2022-12-06:
So hope it will last long :)
Anyway to sum things up. Raspotify works perfectly, but routers are mad disturbers.
Once again thanks a lot for whole your help. For sure I will boder you again if something goes wrong :)

[32 more replies]

closed 2022-12-07 by jackobtone