dtcooper/raspotify #601: Raspotify vs MPD/flac

MitraMai, 2022-12-27:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

A quick question, which I think I already know the answer to but I do want to ask it anyway.
When playing the same song on my RPI4 equipped with a IQaudIODAC Pro the sound on Raspotify is much "flatter" than when playing the song in flac- format on MPD. Which, obviously, is much richer and with more nuances to it.
Is it all due to the bitrate of Spotify, or can I somehow "enhance" the audio on Raspotify?
This is my asound.conf, configurated with asound-conf-wizard:

```
defaults.pcm.rate_converter speexrate_medium
pcm.playback {
    type dmix
    ipc_key {
        @func refer
        name defaults.pcm.ipc_key
    }
    ipc_gid {
        @func refer
        name defaults.pcm.ipc_gid
    }
    ipc_perm {
        @func refer
        name defaults.pcm.ipc_perm
    }
    tstamp_type {
        @func refer
        name defaults.pcm.tstamp_type
    }
    slave {
        pcm {
            type hw
            card IQaudIODAC
            device 0
            subdevice 0
        }
        channels 2
        rate 384000
        format S32_LE
        period_size 0
        buffer_size 0
        periods 0
        buffer_time 170000
        period_time 34000
    }
}

pcm.!default {
    type asym
    capture.pcm {
        type plug
        slave.pcm "null"
    }
    playback.pcm {
        type plug
        slave.pcm "playback"
    }
}

ctl.!default {
    type hw
    card IQaudIODAC
```

And the relevant part of the Raspotify.conf:

```
# Bitrate (kbps) {96|160|320}. Defaults to 160.
#LIBRESPOT_BITRATE="160"
LIBRESPOT_BITRATE="320"
# Output format {F64|F32|S32|S24|S24_3|S16}. Defaults to S16.
#LIBRESPOT_FORMAT="S16"
LIBRESPOT_FORMAT="S32"
# Displayed device type. Defaults to speaker.
LIBRESPOT_DEVICE_TYPE="speaker" 
```

And this is the relevant part of my MPD conf:
```
audio_output {
        type            "alsa"
#       name            "ALSA Device"
        name            "IQaudIODAC"
        device          "hw:CARD=IQaudIODAC,DEV=0"
#       mixer_type      "hardware"      # optional
        mixer_device    "hw:IQaudIODAC" # optional
        mixer_control   "Digital"       # optional
        
#       mixer_index     "0"             # optional
#       mixer_type      "software"
}
```

JasonLG1979, 2022-12-27:
Normalization is enabled by default. That generally lowers the over all level a couple dB? That would make things sound less in your face.

As a side note, there's no reason to upsample to 384000. You're just waisting CPU cycles and degrading the audio fidelity.

MitraMai, 2022-12-28:
I played a bit with the values in the Normalization-section but couldn't hear any distinct difference.
May I ask why the higher value in upsampling degrades the audio fidelity?
 Anyhow... Thanks for the reply.

JasonLG1979, 2022-12-28:
Upsampling degrades sound in the same way it degrades the quality of an image if you blow it to 8 times it's size. You're asking an algorithm to guess about information that is not there.

It's pretty common knowledge. You can Google audio upsampling and it's effects on audio fidelity.

closed 2022-12-28 by JasonLG1979

JasonLG1979, 2022-12-30:
For a direct "apples to apples" comparison you should:

1. Completely disable normalization:
```
# Play all tracks at approximately the same apparent volume.
#LIBRESPOT_ENABLE_VOLUME_NORMALISATION=
```

2. Set the volume to 100 and fixed:
```
# Initial volume in % from 0 - 100.
# Defaults to 50 For the alsa mixer: the current volume.
LIBRESPOT_INITIAL_VOLUME="100"

# Volume control scale type {cubic|fixed|linear|log}.
# Defaults to log.
LIBRESPOT_VOLUME_CTRL="fixed"
```

3. Set the format and bitrate to as high as possible, (which you have already done).
```
# Bitrate (kbps) {96|160|320}. Defaults to 160.
LIBRESPOT_BITRATE="320"

# Output format {F64|F32|S32|S24|S24_3|S16}. Defaults to S16.
LIBRESPOT_FORMAT="S32"
```

4. Directly address the DAC, bypassing any sound processing.

```
# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
LIBRESPOT_DEVICE="hw:CARD=IQaudIODAC,DEV=0"
```

That will provide the closest thing to "bitPerfect" as you can get given that Spotify streams lossy audio.

Be aware though that since both librespot and MPD are both directly addressing the physical device trying to play one while another is playing with cause either or both to crash since only one thing can access the physical card at a time.

MitraMai, 2022-12-31:
Oh yes!
The quality is much better! Only problem is that the configuration also fixes the volume. I have an old NAD- amplifier without remote, so I guess I'll have to think about which conf is preferable.
Thank you very much for the explanation and a happy new year!

JasonLG1979, 2023-01-01:
> Only problem is that the configuration also fixes the volume

That part is completely optional. The default volume scaling is logarithmic. I would also try cubic and see which you prefer.

MitraMai, 2023-01-06:
Yup, that works too!
Much better now. Thanks again!