dtcooper/raspotify #604: raspotify keeps crashes when trying to play a song.

wolfiex, 2023-01-04:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

- Librespot works fine with the 'default' device. 
`librespot --name "librespot test" --backend alsa --cache /tmp/librespot-test --bitrate 160 -v -d default`

- I have tried setting `LIBRESPOT_DEVICE="default"` or `OPTIONS="--device default"` in the config file with no avail. 

- All other apps work fine with the default audio selection. 

- raspotify connects fine, just seems to crash / disconnect each time I press play on a song. 

- default raspotify configuration from the install. 

- No idea why it is searching for PCM bluealsa - this is not specified anywhere

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any

```shell
Jan 04 18:02:05 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Jan 04 18:02:14 raspberrypi librespot[8828]: [2023-01-04T18:02:14Z WARN  librespot_core::apresolve] Ignoring blacklisted acces>
Jan 04 18:02:16 raspberrypi librespot[8828]: Failed to get home directory.
Jan 04 18:02:16 raspberrypi librespot[8828]: ALSA lib pcm.c:2660:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Jan 04 18:02:16 raspberrypi librespot[8828]: [2023-01-04T18:02:16Z ERROR librespot_playback::player] Audio Sink Error Connecti>
Jan 04 18:02:16 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Jan 04 18:02:16 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Jan 04 18:02:16 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
```

wolfiex, 2023-01-04:
Solution found after trying to get raspotify to work. 

```
wget https://github.com/Spotifyd/spotifyd/releases/download/v0.3.4/spotifyd-linux-armv6-slim.tar.gz;
tar zxvf spotifyd-linux-armv6-slim.tar.gz;
./spotifyd - no-daemon
```

Connect from device and play.

closed 2023-01-04 by wolfiex

JasonLG1979, 2023-01-04:
What version of Raspotify and OS are you running. `OPTIONS="--device default"` hasn't worked in a long time and bluealsa isn't in Debian stable's repos anymore?

Bluealsa has to be specified somewhere otherwise it would not be trying to connect to it.