dtcooper/raspotify #605: Raspotify disconnects/drops when Play is pressed on phone

pdpbjorn, 2023-01-18:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

The Phone Spotify sees Raspotify on the network an does connect - icon turns green. But immediately after Play is pressed, the connection is dropped.
I see some error message in the log, but I do not have the knowledge to troubleshoot

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any

```shell
Linux raspberrypi 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

-- Logs --

Jan 17 21:01:52 raspberrypi librespot[1530]: [2023-01-17T20:01:52Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Jan 17 21:01:52 raspberrypi librespot[1530]: [2023-01-17T20:01:52Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Jan 17 21:01:52 raspberrypi librespot[1530]: [2023-01-17T20:01:52Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Jan 17 21:01:52 raspberrypi librespot[1530]: [2023-01-17T20:01:52Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gue1.spotify.com:443
Jan 17 21:02:02 raspberrypi librespot[1530]: Failed to get home directory.
Jan 17 21:02:02 raspberrypi librespot[1530]: [2023-01-17T20:02:02Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device default May be Invalid, Busy, or Already in Use, ALSA function 'snd_pcm_open' failed with error 'UnknownErrno: Unknown errno'
Jan 17 21:02:02 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Jan 17 21:02:02 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Jan 17 21:02:02 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.

-- Config --

LIBRESPOT_QUIET=
LIBRESPOT_AUTOPLAY=
LIBRESPOT_DISABLE_AUDIO_CACHE=
LIBRESPOT_DISABLE_CREDENTIAL_CACHE=
LIBRESPOT_ENABLE_VOLUME_NORMALISATION=

-- Ouput of aplay -l --

**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi [vc4-hdmi], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: IQaudIODAC [IQaudIODAC], device 0: IQaudIO DAC HiFi pcm512x-hifi-0 [IQaudIO DAC HiFi pcm512x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

-- Ouput of aplay -L --

null
    Discard all samples (playback) or generate zero samples (capture)
default
    Default Audio Device
sysdefault
    Default Audio Device
lavrate
    Rate Converter Plugin Using Libav/FFmpeg Library
samplerate
    Rate Converter Plugin Using Samplerate Library
speexrate
    Rate Converter Plugin Using Speex Resampler
jack
    JACK Audio Connection Kit
oss
    Open Sound System
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
vdownmix
    Plugin for channel downmix (stereo) with a simple spacialization
hw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Hardware device with all software conversions
default:CARD=vc4hdmi
    vc4-hdmi, MAI PCM i2s-hifi-0
    Default Audio Device
sysdefault:CARD=vc4hdmi
    vc4-hdmi, MAI PCM i2s-hifi-0
    Default Audio Device
hdmi:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    HDMI Audio Output
dmix:CARD=vc4hdmi,DEV=0
    vc4-hdmi, MAI PCM i2s-hifi-0
    Direct sample mixing device
usbstream:CARD=vc4hdmi
    vc4-hdmi
    USB Stream Output
hw:CARD=IQaudIODAC,DEV=0
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Direct hardware device without any conversions
plughw:CARD=IQaudIODAC,DEV=0
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Hardware device with all software conversions
default:CARD=IQaudIODAC
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Default Audio Device
sysdefault:CARD=IQaudIODAC
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Default Audio Device
dmix:CARD=IQaudIODAC,DEV=0
    IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
    Direct sample mixing device
usbstream:CARD=IQaudIODAC
    IQaudIODAC
    USB Stream Output

-- Ouput of librespot -d ? --

        Compatible alsa device(s):

        ------------------------------------------------------

        Device:

                hw:CARD=IQaudIODAC,DEV=0

        Description:

                IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
                Direct hardware device without any conversions

        Supported Format(s):

                S16 S24 S32

        ------------------------------------------------------

        Device:

                plughw:CARD=IQaudIODAC,DEV=0

        Description:

                IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
                Hardware device with all software conversions

        Supported Format(s):

                S16 S24 S24_3 S32 F32 F64

        ------------------------------------------------------

        Device:

                default:CARD=IQaudIODAC

        Description:

                IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
                Default Audio Device

        Supported Format(s):

                S16 S24 S24_3 S32 F32 F64

        ------------------------------------------------------

        Device:

                sysdefault:CARD=IQaudIODAC

        Description:

                IQaudIODAC, IQaudIO DAC HiFi pcm512x-hifi-0
                Default Audio Device

        Supported Format(s):

                S16 S24 S24_3 S32 F32 F64

        ------------------------------------------------------
```

JasonLG1979, 2023-01-18:
Are you trying to use Raspotify on a desktop system? Raspotify is intended to be used on headless systems only, as clearly stated in the [readme](https://github.com/dtcooper/raspotify#readme).

pdpbjorn, 2023-01-19:
I'm using Raspotify on a RaspberryPi - as clearly stated in first line of the log :-)
It has been working flawlessly for a couple of years - this behavior is new since the last few weeks

Cheers
/per

JasonLG1979, 2023-01-19:
What does your asound.conf look like if there is one?

JasonLG1979, 2023-01-19:
> I'm using Raspotify on a RaspberryPi - as clearly stated in first line of the log :-)

Raspberry Pi OS comes in desktop and headless versions.

pdpbjorn, 2023-01-19:
I have no `asound.conf `in `/etc`, where I suppose it should be and I cannot seem to find it anywhere else. 
I have a folder -` /etc/alsa/conf.d` - with a number of .conf files: 
```
10-rate-lav.conf      50-oss.conf          98-usb-stream.conf
10-samplerate.conf    50-pulseaudio.conf   99-pulseaudio-default.conf.example
10-speexrate.conf     60-a52-encoder.conf  99-pulse.conf
50-arcam-av-ctl.conf  60-upmix.conf
50-jack.conf          60-vdownmix.conf
```
My `/boot/config.txt` is modifyed for Raspotify to work with my sound-card and disable on-board audio:
```
#dtparam=audio=on
dtoverlay=iqaudio-dacplus
dtparam=i2s=on
```
Setting these back to default:
```
dtparam=audio=on
#dtoverlay=iqaudio-dacplus
#dtparam=i2s=on
```
will actually let Raspotify work, and output through the RaspberryPi onboard Jack-socket.
There seems to be some solution here, but I would actually prefer to use the sound-card and ist phono connectors
Any possibilities?

JasonLG1979, 2023-01-19:
> I have no asound.conf in /etc, where I suppose it should be and I cannot seem to find it anywhere else.

It doesn't exist by default. It's Ok to not have one.

> #dtparam=audio=on
> dtoverlay=iqaudio-dacplus
> dtparam=i2s=on

Maybe try commenting out the i2s line?

pdpbjorn, 2023-01-19:
#dtparam=i2s=on
Unfortunately no luck - the connection still drops.
Log:
```
Jan 19 22:45:59 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Jan 19 22:47:01 raspberrypi librespot[525]: [2023-01-19T21:47:01Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Jan 19 22:47:01 raspberrypi librespot[525]: [2023-01-19T21:47:01Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Jan 19 22:47:01 raspberrypi librespot[525]: [2023-01-19T21:47:01Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Jan 19 22:47:05 raspberrypi librespot[525]: Failed to get home directory.
Jan 19 22:47:05 raspberrypi librespot[525]: [2023-01-19T21:47:05Z ERROR librespot_playback::player] Audio Sink Error Connection Refused: <AlsaSink> Device d>
Jan 19 22:47:05 raspberrypi systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Jan 19 22:47:05 raspberrypi systemd[1]: raspotify.service: Failed with result 'exit-code'.
Jan 19 22:47:05 raspberrypi systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
Jan 19 22:47:18 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).

```

JasonLG1979, 2023-01-19:
> Jan 19 22:47:05 raspberrypi librespot[525]: Failed to get home directory.

This is a very strange line. There's no reason the Raspotify service should be looking for `/home`. It's ran under a dynamic user that has no `home` and it's locked down so that it can not access `/home`. The only folders/files the service can see are it's own. It has basically no access to the file system.

pdpbjorn, 2023-01-20:
The line still appears in log, when raspotify use internal sound, but not the following line on `Audio Sink Error`:
`dtparam=audio=on`
```
-- Journal begins at Mon 2022-04-04 14:17:02 CEST, ends at Fri 2023-01-20 08:59>
Jan 20 08:55:24 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Clie>
Jan 20 08:58:09 raspberrypi librespot[502]: [2023-01-20T07:58:09Z WARN  libresp>
Jan 20 08:58:09 raspberrypi librespot[502]: [2023-01-20T07:58:09Z WARN  libresp>
Jan 20 08:58:09 raspberrypi librespot[502]: [2023-01-20T07:58:09Z WARN  libresp>
Jan 20 08:58:09 raspberrypi librespot[502]: [2023-01-20T07:58:09Z WARN  libresp>
Jan 20 08:58:16 raspberrypi librespot[502]: [2023-01-20T07:58:16Z WARN  libresp>
Jan 20 08:58:19 raspberrypi librespot[502]: Failed to get home directory.
```

pdpbjorn, 2023-01-20:
Enabling both sound outputs at the same time,
```
dtparam=audio=on
dtoverlay=iqaudio-dacplus
dtparam=i2s=on
```
seems to work
```
 $ cat /proc/asound/modules
 0 snd_bcm2835
 1 vc4
 2 snd_soc_iqaudio_dac
```

I can produce sound on 0 and 2, using these:
` $ speaker-test -c2 -twav -l7 -D plughw:0,0`
` $ speaker-test -c2 -twav -l7 -D plughw:2,0`

1 will fail:
` $ speaker-test -c2 -twav -l7 -D plughw:1,0`
```
speaker-test 1.2.4

Playback device is plughw:1,0
Stream parameters are 48000Hz, S16_LE, 2 channels
WAV file(s)
Playback open error: -524,Unknown error 524

```

Raspotify still works; producing sound on the build in device ( 0 snd_bcm2835):

Can I somehow direct Raspotify to use  `2 snd_soc_iqaudio_dac`?

JasonLG1979, 2023-01-20:
Looks like maybe your dac hat is faulty?

JasonLG1979, 2023-01-20:
> Can I somehow direct Raspotify to use `2 snd_soc_iqaudio_dac`?

If your DAC hat supports 44100 you can tell librespot to address it directly by changing:
```
# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
#LIBRESPOT_DEVICE="default"
```
To:
```
# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
LIBRESPOT_DEVICE="hw:CARD=IQaudIODAC,DEV=0"
```

In `/etc/raspotify/conf`.

Be sure to restart the Raspotify service with `sudo systemctl restart raspotify`

Do note that only one program at a time can access the raw device at a time.

netlest, 2023-01-22:
You can disable all other audio cards on Raspberry to have only IQAudioDAC as default device.

Add below to /boot/config.txt
dtoverlay=vc4-kms-v3d,noaudio
Ensure below is commented
#dtparam=audio=on

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: IQaudIODAC [IQaudIODAC], device 0: IQaudIO DAC HiFi pcm512x-hifi-0 [IQaudIO DAC HiFi pcm512x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $

JasonLG1979, 2023-01-22:
I know this but clearly that's not working correctly for some reason.

pdpbjorn, 2023-01-23:
> ```
> # Audio device to use, use `librespot --device ?` to list options.
> # Defaults to the system's default.
> LIBRESPOT_DEVICE="hw:CARD=IQaudIODAC,DEV=0"
> ```
>  
YES! That did it!
Thank you so much!!!

JasonLG1979, 2023-01-23:
You're very welcome. Glad I could help.

closed 2023-01-23 by JasonLG1979