dtcooper/raspotify #606: What's the best sample Rate converter?

fjromerom, 2023-01-20:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi,

I've installed raspotify in a Raspberry PI model 3 and it works great. However when making the configuration I was not sure about what sample rate converter choose.

I've looked for information about every converter listed but I did not find any documentation about them. Although I've tried a couple of them and I don't hear any difference, I'd like to know what's the best recommendation.

Thanks,
Francisco

JasonLG1979, 2023-01-20:
If you can't tell the difference then it really doesn't matter does it? But purely from a technical standpoint `speexrate_medium` is probably the best bang for your buck. It provides a good balance of sound quality to CPU usage.

JasonLG1979, 2023-01-20:
As far as documentation:

The `samplerate*` converters use [libsamplerate (Secret Rabbit Code)](http://www.mega-nerd.com/SRC/)

The `speexrate*` converters use [the speex resampler](https://speex.org/docs/manual/speex-manual/node7.html#SECTION00760000000000000000)

The `lavrate*` converters use [the libav (FFmpeg) resampler](https://github.com/libav/libav/blob/master/libavresample/resample.c)

They're all pretty comparable to my ears sound quality-wise. Generally speaking though samplerate conversion should be avoided if possible, as it will never make things sound better no matter how good the converter is. You for example would not want to configure your device to it's max supported sampling rate for no other reason then it's the max supported sampling rate. If your device supports 44100 audio and you mostly play 44100 audio (Librespot outputs 44100) You would want to configure your device for 44100 so that there is no need for sample rate conversion.

fjromerom, 2023-02-12:
Alright, thanks for the info @JasonLG1979.

closed 2023-02-12 by fjromerom