dtcooper/raspotify #609: Librespot panics and fills syslog and daemon.log till SD capacity

pililac, 2023-03-03:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

The both files /var/log/syslog and /var/log/daemon.log are filled till SD capacity is reached by librespot logs. Raspi stops working due to capacity limit reached.

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
pi@raspberry:~ $ sudo journalctl -u raspotify -b
Journal file /var/log/journal/4cc15c5c33bd417484c266a98199caf9/user-1000@0005f601c8747831-d5fce9b1e92d0ff3.journal~ is truncated, ignoring file.
-- Journal begins at Sat 2023-02-25 08:56:02 GMT, ends at Fri 2023-03-03 17:23:05 GMT. --
Feb 25 08:56:02 raspberry librespot[475]: 75668705:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668706:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668707:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668708:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668709:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668710:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668711:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668712:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668713:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668714:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668715:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668716:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668717:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668718:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668719:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668720:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668721:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668722:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668723:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668724:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668725:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668726:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668727:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668728:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668729:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668730:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668731:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668732:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668733:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668734:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668735:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
Feb 25 08:56:24 raspberry librespot[475]: 75668736:       0x557167a9b4 - std::sys_common::backtrace::__rust_end_short_backtrace::h45ae12a4d7295b52
[... 263 lines cut ...]

                bcm2835 Headphones, bcm2835 Headphones
                Default Audio Device

        Supported Format(s):

                S16 S24 S24_3 S32 F32 F64

        ------------------------------------------------------

        Device:

                dmix:CARD=Headphones,DEV=0

        Description:

                bcm2835 Headphones, bcm2835 Headphones
                Direct sample mixing device

        Supported Format(s):

                S16

        ------------------------------------------------------
```

pililac, 2023-03-03:
Sorry, it was NOT the newest version of raspotify, I'm using 0.43.29

````
pi@raspberry:~ $ apt show raspotify
Package: raspotify
Version: 0.43.29~librespot.v0.4.2-5cda505
Priority: optional
Section: base
Maintainer: Jason Gray <jasonlevigray3@gmail.com>
Installed-Size: 7,732 kB
Depends: libc6 (>= 2.31), coreutils (>= 8.32), libasound2 (>= 1.2.4), alsa-utils (>= 1.2.4), libpulse0 (>= 14.2), systemd (>= 247.3), init-system-helpers (>= 1.60)
Homepage: https://github.com/dtcooper/raspotify
Vcs-Browser: https://github.com/dtcooper/raspotify
Vcs-Git: https://github.com/dtcooper/raspotify
Download-Size: 1,674 kB
APT-Manual-Installed: yes
APT-Sources: https://dtcooper.github.io/raspotify raspotify/main arm64 Packages
Description: Raspotify
 A Spotify Connect systemd daemon for Debian Stable based headless systems which thinly wraps a fork of librespot.
 [Built with rustc 1.67.1]
````

JasonLG1979, 2023-03-03:
Very strange? `/etc/raspotify/crash_report` mentions `/tmp/.tmpimGKVk` is not found. That possibly refers to a track that is supposed to cached to disk? After you clear the logs does the problem persist? `/etc/raspotify/crash_report` ***should*** capture the most recent relevant logs. I suppose if all else fails after a hard failure and log capture I could clear the logs?

JasonLG1979, 2023-03-03:
Well never mind, my bad turns out you can't delete just one service's logs since the logs are interlaced. Kinda stupid design choice by systemd if you ask me...

JasonLG1979, 2023-03-03:
The error seems to be coming from [this line](https://github.com/JasonLG1979/librespot/blob/raspotify/audio/src/fetch/mod.rs#L365) in the meantime until we figure out the cause I can make librespot cleanly exit with an error message and not puke a huge backtrace at least. Not ideal but better then what it's doing now.

JasonLG1979, 2023-03-09:
@pililac I pushed this [commit](https://github.com/JasonLG1979/librespot/commit/b03804e4ceb4f75a27e81703575695270724c958) that causes librespot to exit more cleanly.

Can you test the latest Raspotify release?

JasonLG1979, 2023-03-20:
I can not reproduce this bug. Closing this because of a lack of response.

closed 2023-03-20 by JasonLG1979