dtcooper/raspotify #611: `raspotify.service` is not starting “[… ERROR librespot] `--username` / `-u` can not be an empty string”

fooness, 2023-03-06:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hello everyone.

Unfortunately, I’m having problems with getting `raspotify.service` started since after the breaking changes.

I uninstalled raspotify via `apt purge raspotify` and installed again via `apt install raspotify`, as stated in the wiki.

Then, I slightly modified the new config, i.e. changed the device name, and increased the bitrate and format.

The default config [clearly states](https://github.com/dtcooper/raspotify/blob/master/raspotify/etc/raspotify/conf#L76): “Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.”

```
$ sudo systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: inactive (dead) since Mon 2023-03-06 19:04:04 CET; 3s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
    Process: 9739 ExecStart=/usr/bin/librespot (code=exited, status=1/FAILURE)
   Main PID: 9739 (code=exited, status=1/FAILURE)
        CPU: 109ms

Mar 06 19:04:04 raspberry systemd[1]: Started Raspotify (Spotify Connect Client).
Mar 06 19:04:04 raspberry librespot[9739]: [2023-03-06T18:04:04Z ERROR librespot] `--username` / `-u` can not be an empty string
Mar 06 19:04:04 raspberry systemd[1]: raspotify.service: Main process exited, code=exited, status=1/FAILURE
Mar 06 19:04:04 raspberry systemd[1]: raspotify.service: Failed with result 'exit-code'.
Mar 06 19:04:04 raspberry systemd[1]: raspotify.service: Triggering OnFailure= dependencies.
```

Thank you in advance for your advice and help.

JasonLG1979, 2023-03-06:
Post your config.

fooness, 2023-03-07:
```
$ ls -alF /etc/raspotify
drw-------  2 root root  4096 Mar  7 08:37 ./
-rw-------  1 root root  4579 Mar  6 19:16 conf
```

```
$ cat /etc/raspotify/conf
# /etc/raspotify/conf -- Arguments/configuration for librespot

# A non-exhaustive list of librespot options and flags.

# Please see https://github.com/dtcooper/raspotify/wiki &
# https://github.com/librespot-org/librespot/wiki/Options
# for configuration details and a full list of options and flags.

# You can also find a full list with `librespot -h`.

# To avoid name collisions environment variables must be prepended with
# `LIBRESPOT_`, so option/flag `foo-bar` becomes `LIBRESPOT_FOO_BAR`.

# Invalid environment variables will be ignored.

# Raspotify defaults may vary from librespot defaults.
# Commenting out the environment variable will fallback to librespot's default
# unless otherwise noted.

# Flags are either on (uncommented) or off (commented),
# their values are otherwise not evaluated (but the "=" is still needed).

# Only log warning and error messages.
LIBRESPOT_QUIET=

# Automatically play similar songs when your music ends.
#LIBRESPOT_AUTOPLAY=

# Disable caching of the audio data.
# Enabling audio data caching can take up a lot of space
# if you don't limit the cache size with LIBRESPOT_CACHE_SIZE_LIMIT.
# It can also wear out your Micro SD card. You have been warned.
LIBRESPOT_DISABLE_AUDIO_CACHE=

# Disable caching of credentials.
# Caching of credentials is not necessary so long as
# LIBRESPOT_DISABLE_DISCOVERY is not set.
LIBRESPOT_DISABLE_CREDENTIAL_CACHE=

# Play all tracks at approximately the same apparent volume.
LIBRESPOT_ENABLE_VOLUME_NORMALISATION=

# Enable verbose log output.
#LIBRESPOT_VERBOSE=

# Disable zeroconf discovery mode.
#LIBRESPOT_DISABLE_DISCOVERY=

# Options will fallback to their defaults if commented out,
# otherwise they must have a valid value.

# Device name.
# Raspotify defaults to "raspotify (*hostname)".
# Librespot defaults to "Librespot".
LIBRESPOT_NAME="raspberry"

# Bitrate (kbps) {96|160|320}. Defaults to 160.
LIBRESPOT_BITRATE="320"

# Output format {F64|F32|S32|S24|S24_3|S16}. Defaults to S16.
LIBRESPOT_FORMAT="S32"

# Displayed device type. Defaults to speaker.
LIBRESPOT_DEVICE_TYPE="speaker"

# Limits the size of the cache for audio files.
# It's possible to use suffixes like K, M or G, e.g. 16G for example.
# Highly advised if audio caching isn't disabled. Otherwise the cache
# size is only limited by disk space.
#LIBRESPOT_CACHE_SIZE_LIMIT=""

# Audio backend to use, alsa or pulseaudio. Defaults to alsa.
#LIBRESPOT_BACKEND="alsa"

# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
LIBRESPOT_USERNAME=""

# Password used to sign in with.
LIBRESPOT_PASSWORD=""

# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
#LIBRESPOT_DEVICE="default"

# Initial volume in % from 0 - 100.
# Defaults to 50 For the alsa mixer: the current volume.
LIBRESPOT_INITIAL_VOLUME="10"

# Volume control scale type {cubic|fixed|linear|log}.
# Defaults to log.
LIBRESPOT_VOLUME_CTRL="log"
#LIBRESPOT_VOLUME_CTRL="linear"

# Range of the volume control (dB) from 0.0 to 100.0.
# Default for softvol: 60.0.
# For the alsa mixer: what the control supports.
#LIBRESPOT_VOLUME_RANGE="60.0"

# Pregain (dB) applied by volume normalisation from -10.0 to 10.0.
# Defaults to 0.0.
#LIBRESPOT_NORMALISATION_PREGAIN="0.0"

# Threshold (dBFS) at which point the dynamic limiter engages
# to prevent clipping from 0.0 to -10.0.
# Defaults to -2.0.
#LIBRESPOT_NORMALISATION_THRESHOLD="-2.0"

# The port the internal server advertises over zeroconf 1 - 65535.
# Ports <= 1024 may require root privileges.
#LIBRESPOT_ZEROCONF_PORT=""

# HTTP proxy to use when connecting.
#LIBRESPOT_PROXY=""

# ### This is NOT a librespot option or flag. ###
# This modifies the behavior of the Raspotify service.
# If you have issues with this option DO NOT file a bug with librespot.
#
# By default librespot "download buffers" tracks, meaning that it downloads
# the tracks to disk and plays them from the disk and then deletes them when
# the track is over. This practice is very common, many other audio frameworks
# and players do the exact same thing as a disk based tmp cache is easy to use
# and very resilient. That being said there may be cases where a user may want
# to minimize disk read/writes.
#
# Commenting this out will cause librespot to use a tmpfs so that provided there
# is enough RAM to hold the track nothing is written to disk but instead to a tmpfs.
# See https://github.com/dtcooper/raspotify/discussions/567
# And https://www.kernel.org/doc/html/latest/filesystems/tmpfs.html
#TMPDIR=/tmp
```

```
$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 [HiFiBerry DAC+ HiFi pcm512x-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

```
$ aplay -Dhw:0 --dump-hw-params /usr/share/sounds/alsa/Front_Right.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Right.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
HW Params of device "hw:0":
--------------------
ACCESS:  MMAP_INTERLEAVED RW_INTERLEAVED
FORMAT:  S16_LE S24_LE S32_LE
SUBFORMAT:  STD
SAMPLE_BITS: [16 32]
FRAME_BITS: [32 64]
CHANNELS: 2
RATE: [8000 384000]
PERIOD_TIME: (83 8192000]
PERIOD_SIZE: [32 65536]
PERIOD_BYTES: [256 524288]
PERIODS: [2 4096]
BUFFER_TIME: (166 16384000]
BUFFER_SIZE: [64 131072]
BUFFER_BYTES: [256 524288]
TICK_TIME: ALL
--------------------
aplay: set_params:1349: Channels count non available
```

JasonLG1979, 2023-03-07:
```
# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
LIBRESPOT_USERNAME=""

# Password used to sign in with.
LIBRESPOT_PASSWORD=""
```

As I suspected. These need to be commented out (their default state) or you need to provide credentials. They can't be empty if present.

closed 2023-03-07 by JasonLG1979

fooness, 2023-03-07:
@JasonLG1979 Thank you for your advice. I absolutely cannot remember modifying/un-commenting these two lines, but it obviously must have happened somehow. I am sorry for this, and very much looking forward to fixing this tonight!

JasonLG1979, 2023-03-07:
It's all good.