dtcooper/raspotify #619: approx 10-15s delay

Blackspell01, 2023-04-06:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hey, first of all thank you for the great work!
My question would be wether a 10 second delay from playing a song till it actually starts is normal?
This topic (https://github.com/dtcooper/raspotify/issues/103) which was closed due to inactivity, dealt with this issue but the problem wasn't solved.
Maybe there is a tweak to reduce the latency or its just a general thing I have to accept.
By the way, download speed of the pi are 50mbits down and 10mbits up. Not overwhelmingly great but should suffice shouldn't it?
Thanks in advance!

JasonLG1979, 2023-04-06:
I don't know what to tell you? I can't make your internet speed faster?

Blackspell01, 2023-04-06:
The internet speed isn't the problem?

JasonLG1979, 2023-04-06:
Do the logs say anything? When you run librespot in debug mode what does it say?

Blackspell01, 2023-04-06:
Apr 06 13:51:23 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z INFO  librespot] librespot 0.4.2 9a92196 (Built on 2023-04-02, Build ID: oQtbI6jK, Profile: release)
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot] Environment variable(s):
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_NAME="Simons Raspberry Pi"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_BACKEND="alsa"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_CACHE="/var/cache/raspotify"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_SYSTEM_CACHE="/var/lib/raspotify"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_VERBOSE="1"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_BITRATE="160"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_CACHE_SIZE_LIMIT="1G"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_DEVICE="default"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot]                 LIBRESPOT_VOLUME_CTRL="linear"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z TRACE librespot] Using cached credentials.
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_discovery::server] Zeroconf server listening on 0.0.0.0:46699
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_discovery::server] Shutting down discovery server
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z INFO  librespot_core::session] Connecting to AP "ap-gew1.spotify.com:4070"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z INFO  librespot_core::session] Authenticated as "nervensäge44" !
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_core::session] new Session[0]
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z INFO  librespot_playback::mixer::softmixer] Mixing with softvol and volume control: Linear
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_connect::spirc] new Spirc[0]
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_connect::spirc] canonical_username: nervensäge44
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot::component] new MercuryManager
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_playback::mixer::mappings] Input volume 25373 mapped to: 38.72%
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_playback::player] new Player[0]
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z INFO  librespot_playback::convert] Converting with ditherer: tpdf
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z INFO  librespot_playback::audio_backend::alsa] Using AlsaSink with format: S16
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_playback::player] command=AddEventSender
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_playback::player] command=VolumeSet(25373)
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_core::session] Session[0] strong=3 weak=2
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z INFO  librespot_core::session] Country: "DE"
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_core::mercury] subscribed uri=hm://remote/user/nervens%C3%A4ge44/ count=0
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_connect::spirc] kMessageTypeNotify "FireTV 4K Wohnzimmer" 05d0272c5c24aa56f9da626eabad661e60ca75b0 1449672607 1680781884561 kPlayStatusStop
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_connect::spirc] kMessageTypeNotify "Mac mini von Simon" d8cd854dd6543d713e6d197b6857ab6392704f9f 1449672607 1680781884561 kPlayStatusStop
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_connect::spirc] kMessageTypeNotify "VSX-830 B0245D" 3846a059dce7272d5f25608de1721c668d4b7c22 1449672607 1680781884561 kPlayStatusStop
Apr 06 13:51:24 raspberrypi librespot[4504]: [2023-04-06T11:51:24Z DEBUG librespot_connect::spirc] kMessageTypeNotify "Simons FireTV 4k" 123345879b0dfeb969253cfaf4bd91f3de5a5a36 1449672607 1680781884561 kPlayStatusStop
[... cut ...]
Apr 06 13:52:20 raspberrypi librespot[4504]: [2023-04-06T11:52:20Z TRACE librespot_playback::audio_backend::alsa] A smaller than necessary Period Size relative to Buffer Size can lead to high CPU usage.
Apr 06 13:52:20 raspberrypi librespot[4504]: [2023-04-06T11:52:20Z TRACE librespot_playback::audio_backend::alsa] Period Buffer size in bytes: 1776
Apr 06 13:52:20 raspberrypi librespot[4504]: [2023-04-06T11:52:20Z TRACE librespot_connect::spirc] Sending status to server: [kPlayStatusPlay]
Apr 06 13:52:20 raspberrypi librespot[4504]: [2023-04-06T11:52:20Z TRACE librespot_connect::spirc] ==> kPlayStatusPlay

Is this the specific log you have requested?

JasonLG1979, 2023-04-06:
That works out to about a 0.74 sec buffer size which is not horrible. So you looking at less than a sec of audio latency. So your issue is network latency somewhere.

JasonLG1979, 2023-04-06:
The currently librespot does not use a CDN so who knows from where the song is downloaded for playback or how long it will take for commands to be issued. That's really on Spotify. The newer API being worked on in the dev branch is planning on using a CDN so there should be much less latency.

Blackspell01, 2023-04-06:
Hmm ok.
But you can see that from starting sink (13:52:10) to kPlayStatusPlay (13:52:20) there's a 10 second gap in between.
How could I check if there's a network latency issue? 
And the "Error getting Buffer Size, Min Buffer Size must be at least twice the Period Size" is nothing to worry about?

JasonLG1979, 2023-04-06:
The buffer message is just the result of a sanity check failing. It's not good but it's not the end of the world. Basically if that fails the backend will use the device's defaults whatever they may be.

JasonLG1979, 2023-04-06:
The network delay is the issue. I would start by pinging the device you're connected to and then try to ping the AP you're connected to.

Blackspell01, 2023-04-06:
[image]

Ok but everything seems to be quite normal actually

JasonLG1979, 2023-04-06:
Ping the AP.

Blackspell01, 2023-04-06:
Im sorry what is the AP ?

JasonLG1979, 2023-04-06:
ap-gew1.spotify.com in your case it's in the logs. I'm not sure if it respond to the ping or not though?

Blackspell01, 2023-04-06:
[image]

yes it does

JasonLG1979, 2023-04-06:
Very strange. Those are very good numbers too? I'm kinda at a loss here.

Blackspell01, 2023-04-06:
Unfortunate, anyways thank you for your help!

JasonLG1979, 2023-04-06:
No problem. With luck it's just something on Spotify's side and it clears itself up in a couple day?

Blackspell01, 2023-04-08:
Hey I just wanted to let you know that I fixed the issue by installing pulse audio instead of ALSA. Now it works flawlessly!
Thank you

JasonLG1979, 2023-04-08:
> Hey I just wanted to let you know that I fixed the issue by installing pulse audio instead of ALSA. Now it works flawlessly!
> Thank you

Weird. You still have ALSA. PulseAudio sits on top ALSA, you can't not have ALSA. I'm very surprised that it would have any noticeable affect at all as far as latency?

JasonLG1979, 2023-04-08:
If anything PulseAudio should add additional audio latency.

closed 2023-04-08 by JasonLG1979