dtcooper/raspotify #621: Problem during installation: The repository is not signed

sosnus, 2023-04-22:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

Hi, I try install dtcooper/raspotify on fresh installation of Debian 11 on RPi2.
When I execute this script, I see output like this:
```
hal9000@raspberrypi:~ $ sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (7.74.0-1.3+deb11u7).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Get:2 https://dtcooper.github.io/raspotify raspotify InRelease [3,143 B]                
Hit:3 http://archive.raspberrypi.org/debian bullseye InRelease                          
Err:2 https://dtcooper.github.io/raspotify raspotify InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY XXXXXXXXXXXXXXX
Reading package lists... Done
W: GPG error: https://dtcooper.github.io/raspotify raspotify InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 09146F2F7953A455
E: The repository 'https://dtcooper.github.io/raspotify raspotify InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
Maybe I forgot about something?

Why I see:
```
W: GPG error: https://dtcooper.github.io/raspotify raspotify InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 09146F2F7953A455
E: The repository 'https://dtcooper.github.io/raspotify raspotify InRelease' is not signed.
```
Should I do it in less secure way?

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
cd: /etc/raspotify/: No such file or directory
```

JasonLG1979, 2023-04-22:
For some strange reason the key wasn't downloaded in the install process from the looks of things?

It should be at `/usr/share/keyrings/raspotify_key.asc`.

If it's not there you should be able to fix the issue with:

```
curl -sSL https://dtcooper.github.io/raspotify/key.asc | sudo tee /usr/share/keyrings/raspotify_key.asc >/dev/null
```
```
sudo chmod 644 /usr/share/keyrings/raspotify_key.asc
```

sosnus, 2023-04-23:
Sucessfully add key using Your commands:
```
curl -sSL https://dtcooper.github.io/raspotify/key.asc | sudo tee /usr/share/keyrings/raspotify_key.asc >/dev/null
sudo chmod 644 /usr/share/keyrings/raspotify_key.asc
```
But still same problem :/ 
```
hal9000@raspberrypi:~ $ sudo chmod 644 /usr/share/keyrings/raspotify_key.asc
hal9000@raspberrypi:~ $ sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (7.74.0-1.3+deb11u7).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease              
Get:3 https://dtcooper.github.io/raspotify raspotify InRelease [3,143 B]    
Err:3 https://dtcooper.github.io/raspotify raspotify InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 09146F2F7953A455
Reading package lists... Done
W: GPG error: https://dtcooper.github.io/raspotify raspotify InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 09146F2F7953A455
E: The repository 'https://dtcooper.github.io/raspotify raspotify InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
hal9000@raspberrypi:~ $ 
```

sosnus, 2023-04-23:
hmmm, after reading a lot of threads on StackOverflow, I found solution, but I don't know it is safety option or not.
If You want avoid this problem, execute:
```
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 09146F2F7953A455
```
before:
```
sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh
```
Spotify working, so for me everything is ok. What do You mean @JasonLG1979 , should we close this issue?
And of course, thank You for rapid response to question! ;)

JasonLG1979, 2023-04-23:
I'm not sure why it happens but I've used Debian based OS's almost exclusively for over 15 yrs and I can count on 1 hand how many times I've had key issues. I'd call it a fluke. 99.999% of the time there should be no reason to explicitly/manually trust the key, but as far as I can tell it's safe to do so, so long as you trust where the key comes from.

This really isn't a Raspotify bug. We're doing everything right. It's just that for some reason unknown (to me at least) something went sideways. You just happened to be that 0.001%.

It's good info to have though so if anyone else has the same issue I can point them to this bug report.

closed 2023-04-23 by JasonLG1979