dtcooper/raspotify #624: Connection reset (os error 104) on RPi3 with HifiberryDAC+

tofublock, 2023-05-11:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

I have been using Raspotify on a RPi3 with HifiberryDAC+ running OSMC for a long time with absolutely no problems. A few months ago I did an upgrade. If I can reconstruct it correctly, the update was 2022-01-18: raspotify:armhf (0.31.4 librespot.v0.3.1-34-ge5fd7d6, 0.31.6 librespot.v0.3.1-47-gc6e97a7)
The reason I think it was this upgrade is that I remember having to update raspotify config (new location) and create an asound.conf to get it to work again.
Since then, without having changed anything else, I see frequent crashes and service reboots with os error 104 (sometimes with MercuryError, sometimes without).

I have read the wiki, tried to run the asound wizard, fiddled with the asound.conf at various points in time since it started happening (see config below). The thing that stands out to me there is the strange samplerate, but it does work.

Any ideas or help would be highly appreciated (and thank you so much for this project in general - absolutely love it!)

Thank you!

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
Feb 19 21:09:04 osmc librespot[402]: [2023-02-19T20:09:04Z ERROR librespot_playback::player] Unable to load audio item: MercuryError
Feb 19 21:09:04 osmc librespot[402]: [2023-02-19T20:09:04Z ERROR librespot_core::session] Connection reset by peer (os error 104)
Feb 19 21:09:04 osmc librespot[402]: [2023-02-19T20:09:04Z ERROR librespot_connect::spirc] subscription terminated
Feb 19 21:09:05 osmc librespot[402]: [2023-02-19T20:09:05Z WARN  librespot] Spirc shut down unexpectedly
...
Feb 20 21:33:00 osmc librespot[402]: [2023-02-20T20:33:00Z ERROR librespot_core::session] Connection reset by peer (os error 104)
Feb 20 21:33:00 osmc librespot[402]: [2023-02-20T20:33:00Z WARN  librespot_connect::spirc] Cannot flush spirc event sender.

## asound.conf:
`pcm.default_hw {
    type hw
    card sndrpihifiberry
    device 0
    channels 2
    rate 352800
    format S32_LE
}

pcm.dmixer {
    type dmix
    slave.pcm default_hw
    ipc_key {
        @func refer
        name defaults.pcm.ipc_key
    }
    ipc_gid {
        @func refer
        name defaults.pcm.ipc_gid
    }
    ipc_perm {
        @func refer
        name defaults.pcm.ipc_perm
    }
    bindings {
        0 0
        1 1
    }
}

pcm.!default {
    type plug

    slave.pcm dmixer
}`
```

tofublock, 2023-05-11:
I now realise that the frame rates from the aplay output are a range.
I tried with 44100 again, resulting in the error `Audio Sink Error Invalid Parameters: <AlsaSink> Device default Unsupported Sample Rate 44100,  ALSA function 'snd_pcm_hw_params_set_rate' failed with error 'EINVAL: Invalid argument', Supported Sample Rate(s): [8000, 11025, 16000, 22050, 48000, 88200, 96000, 176400, 192000, 352800, 384000, 705600, 768000]`
I tried with 48000 - no output. I tried with 88200 and it works. I tried with 48000 yet again and now it does work so far (verdict still out, since the crashes were erratic.)

I'm closing this for now. Sorry for the trouble. Maybe it will help somebody else.

closed 2023-05-11 by tofublock

JasonLG1979, 2023-05-11:
A 104 error is a network error. It has nothing to do with ALSA or your audio config

tofublock, 2023-05-11:
> A 104 error is a network error. It has nothing to do with ALSA or your audio config

You could have said the exact same thing in reply to my comment over in the librespot repo.
Either way, I didn't have the problem until I upgraded. And you're right, the issue persists.

JasonLG1979, 2023-05-11:
You gave no details really at all. How was I supposed to know? That's why you file a bug report here 1st (with info like you did) and then we can get it sorted.

You had 2 problems. 1 the 104 for which nothing really done about. And 2 the audio issue that you sorted out yourself.

tofublock, 2023-05-11:
I didn't really have an audio issue, but my config is arguably better now. Thanks for taking a look. I appreciate your time.
I've rolled back to 0.31.3 too to see if going back to that state helps, but the problem still persists. Curious, it used to work so well. Must have been something else I updated along the way.