dtcooper/raspotify #625: Puling Additional Metadata in onevent such as title or url for album art

txoof, 2023-05-13:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

I have a toy version of the python --onevent script running following the [example in the wiki](https://github.com/librespot-org/librespot/blob/dev/contrib/event_handler_example.py). 

```
{
    "event_time": "2023-05-13 21:58:40.235850",
    "event": "playing",
    "track_id": "36c4JohayB9qd64eidQMBi",
    "position_ms": "0"
}
```

I can successfully pull the event, track_id, position_ms values using this example. 

I can't get the Title or the Album information without resorting to pulling data from the API. I see in `/var/log/syslog` that librespot is fully aware of at least some of this data:

```
[2023-05-13T19:53:35Z INFO  librespot_playback::player] <Fragments of Time (feat. Todd Edwards)> (279773 ms) loaded
```

Is there a way to get this passed into the metadata fields?

The example python script is built to accept these items, but for some reason I'm not seeing this data:
```
    if item_type == 'Track':
        track_metadata_fields = {}
        track_metadata_fields['number'] = os.environ['NUMBER']
        track_metadata_fields['disc_number'] = os.environ['DISC_NUMBER']
        track_metadata_fields['popularity'] = os.environ['POPULARITY']
        track_metadata_fields['album'] = os.environ['ALBUM']
        track_metadata_fields['artists'] = os.environ['ARTISTS'].split(null)
        track_metadata_fields['album_artists'] = os.environ['ALBUM_ARTISTS'].split(null)
        json_dict['track_metadata_fields'] = track_metadata_fields
```

I'm using the example script from `/usr/bin/onefiile.py` and the permissions are set so that librespot can execute the script successfully. I get output from the script in JSON format, but it only inludes the event time, event, track_id and position_ms.

JasonLG1979, 2023-05-13:
The metadata stuff hasn't landed in librespot stable yet.

txoof, 2023-05-14:
Interesting that it's available for the logging output, but not hooked up to the onevent handler.

Is there a dev version, or an alternative that has this? I was using the Java version and using the http interface to get metadata, but that version crashes under the latest Raspbian release.

JasonLG1979, 2023-05-14:
> Interesting that it's available for the logging output, but not hooked up to the onevent handler

That's just the way it is in the stable branch for whatever reason. But not near the amount of metadata is actually available in stable either.

The dev branch uses a newer API that exposes much more. That's how I was able to wire it up to the event handler in dev.

> Is there a dev version, or an alternative that has this?

There's no version of Raspotify that uses the dev branch. You'll have to wait for it to land in stable or your have to build your own librespot binary from the dev branch.

txoof, 2023-05-14:
Thanks for the pointers. I'm going to try to build the dev version myself and see if I can't get to the data I need.

Which librespot repo would be the most logical to start with? Your Raspotify/dev branch or the Librespot/dev branch?

txoof, 2023-05-14:
I've successfully built librespot against the dev branch and it does **exactly** what I need. This is fantastic!

Would you consider adding a dev branch of raspotify that includes the dev version of librespot?  I'd be happy to help test and give feedback if that's helpful. 

I can probably sort this out myself, but this feels rather silly to rebuild this excellent wheel.

txoof, 2023-05-14:
I'm going to close this issue and move it over to the discussions.

closed 2023-05-14 by txoof