dtcooper/raspotify #628: Raspotify not showing up in Spotify device list

eranschweitzer, 2023-06-05:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

I'm on a rasbperry pi 4. I installed raspotify but it is not showing up in my device list.
I see that issue #180 is on the same subject but I don't actually see the answer in the discussion.

* I don't have any firewall (e.g. `ufw` is not installed so [this section](https://github.com/dtcooper/raspotify/wiki/Basic-Setup-Guide#ufw-settings) doesn't apply)
* I haven't changed the configuration (listed below for completeness)
* The service appears to be running:
```bash
pi@raspberrypi:~ $ sudo systemctl status raspotify
● raspotify.service - Raspotify (Spotify Connect Client)
     Loaded: loaded (/lib/systemd/system/raspotify.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2023-06-04 16:58:54 PDT; 31s ago
       Docs: https://github.com/dtcooper/raspotify
             https://github.com/librespot-org/librespot
             https://github.com/dtcooper/raspotify/wiki
             https://github.com/librespot-org/librespot/wiki/Options
   Main PID: 4653 (librespot)
      Tasks: 1 (limit: 3933)
        CPU: 127ms
     CGroup: /system.slice/raspotify.service
             └─4653 /usr/bin/librespot

Jun 04 16:58:54 raspberrypi systemd[1]: Started Raspotify (Spotify Connect Client).
```

What am I doing wrong?

## configuration file
```bash
# A non-exhaustive list of librespot options and flags.

# Please see https://github.com/dtcooper/raspotify/wiki &
# https://github.com/librespot-org/librespot/wiki/Options
# for configuration details and a full list of options and flags.

# You can also find a full list with `librespot -h`.

# To avoid name collisions environment variables must be prepended with
# `LIBRESPOT_`, so option/flag `foo-bar` becomes `LIBRESPOT_FOO_BAR`.

# Invalid environment variables will be ignored.

# Raspotify defaults may vary from librespot defaults.
# Commenting out the environment variable will fallback to librespot's default
# unless otherwise noted.

# Flags are either on (uncommented) or off (commented),
# their values are otherwise not evaluated (but the "=" is still needed).

# Only log warning and error messages.
LIBRESPOT_QUIET=

# Automatically play similar songs when your music ends.
LIBRESPOT_AUTOPLAY=

# Disable caching of the audio data.
# Enabling audio data caching can take up a lot of space
# if you don't limit the cache size with LIBRESPOT_CACHE_SIZE_LIMIT.
# It can also wear out your Micro SD card. You have been warned.
LIBRESPOT_DISABLE_AUDIO_CACHE=

# Disable caching of credentials.
# Caching of credentials is not necessary so long as
# LIBRESPOT_DISABLE_DISCOVERY is not set.
LIBRESPOT_DISABLE_CREDENTIAL_CACHE=

# Play all tracks at approximately the same apparent volume.
LIBRESPOT_ENABLE_VOLUME_NORMALISATION=

# Enable verbose log output.
#LIBRESPOT_VERBOSE=

# Disable zeroconf discovery mode.
#LIBRESPOT_DISABLE_DISCOVERY=

# Options will fallback to their defaults if commented out,
# otherwise they must have a valid value.

# Device name.
# Raspotify defaults to "raspotify (*hostname)".
# Librespot defaults to "Librespot".
#LIBRESPOT_NAME="Librespot"

# Bitrate (kbps) {96|160|320}. Defaults to 160.
#LIBRESPOT_BITRATE="160"

# Output format {F64|F32|S32|S24|S24_3|S16}. Defaults to S16.
#LIBRESPOT_FORMAT="S16"

# Displayed device type. Defaults to speaker.
#LIBRESPOT_DEVICE_TYPE="speaker"

# Limits the size of the cache for audio files.
# It's possible to use suffixes like K, M or G, e.g. 16G for example.
# Highly advised if audio caching isn't disabled. Otherwise the cache
# size is only limited by disk space.
#LIBRESPOT_CACHE_SIZE_LIMIT=""

# Audio backend to use, alsa or pulseaudio. Defaults to alsa.
#LIBRESPOT_BACKEND="alsa"

# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
#LIBRESPOT_USERNAME=""

# Password used to sign in with.
#LIBRESPOT_PASSWORD=""

# Audio device to use, use `librespot --device ?` to list options.
# Defaults to the system's default.
#LIBRESPOT_DEVICE="default"

# Initial volume in % from 0 - 100.
# Defaults to 50 For the alsa mixer: the current volume.
#LIBRESPOT_INITIAL_VOLUME="50"

# Volume control scale type {cubic|fixed|linear|log}.
# Defaults to log.
#LIBRESPOT_VOLUME_CTRL="log"

# Range of the volume control (dB) from 0.0 to 100.0.
# Default for softvol: 60.0.
# For the alsa mixer: what the control supports.
#LIBRESPOT_VOLUME_RANGE="60.0"

# Pregain (dB) applied by volume normalisation from -10.0 to 10.0.
# Defaults to 0.0.
#LIBRESPOT_NORMALISATION_PREGAIN="0.0"

# Threshold (dBFS) at which point the dynamic limiter engages
# to prevent clipping from 0.0 to -10.0.
# Defaults to -2.0.
#LIBRESPOT_NORMALISATION_THRESHOLD="-2.0"

# The port the internal server advertises over zeroconf 1 - 65535.
# Ports <= 1024 may require root privileges.
#LIBRESPOT_ZEROCONF_PORT=""

# HTTP proxy to use when connecting.
#LIBRESPOT_PROXY=""

# ### This is NOT a librespot option or flag. ###
# This modifies the behavior of the Raspotify service.
# If you have issues with this option DO NOT file a bug with librespot.
#
# By default librespot "download buffers" tracks, meaning that it downloads
# the tracks to disk and plays them from the disk and then deletes them when
# the track is over. This practice is very common, many other audio frameworks
# and players do the exact same thing as a disk based tmp cache is easy to use
# and very resilient. That being said there may be cases where a user may want
# to minimize disk read/writes.
#
# Commenting this out will cause librespot to use a tmpfs so that provided there
# is enough RAM to hold the track nothing is written to disk but instead to a tmpfs.
# See https://github.com/dtcooper/raspotify/discussions/567
# And https://www.kernel.org/doc/html/latest/filesystems/tmpfs.html
TMPDIR=/tmp
```

eranschweitzer, 2023-06-05:
##Update
When I put in my credentials then raspotify shows up. 
So I'm changing the title since this is just means that it's confusing that the `conf` file states that no credentials are needed for network discovery.

here:
```
# Username used to sign in with.
# Credentials are not required if LIBRESPOT_DISABLE_DISCOVERY is not set.
#LIBRESPOT_USERNAME=""
```

and here:
```
# Password used to sign in with.
#LIBRESPOT_PASSWORD=""
```

rsm86, 2023-06-05:
Commenting to say I'm having the same issue. Haven't tried inputting my credenditals in the conf file (I'd like to avoid this if possible as a. I don't like putting up my credentials in a non-encrypted file, and b. Seems counter productive to the fact that rpi is wired to a speaker where I expect different people to be able to use spotify connect with, I'm not the sole user).

Raspotify has been working great for me for a long time now. I noticed this started happening yesterday. I'll try a reinstall and report back.

eranschweitzer, 2023-06-17:
UPDATE:
It *does* seem to be possible to connect with a different account then the one whose credentials are stored in the config file

*However*

I'm still experiencing very unstable operation. Even though I can see that the raspotify service is running, it is not consistently showing up on the device list in Spotify.

JasonLG1979, 2023-06-26:
ZeroConf can be very finicky if your network devices do not implement it correctly. For example my wifi router flat out refused to work. I ended up getting a dedicated router and using my wifi router as an AP.

closed 2023-10-02 by github-actions[bot]