dtcooper/raspotify #629: Multi Channels

EG-Julien, 2023-06-09:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi,
I'm wondering if it's possible to have play the sound on more than two channels ? Is it related to Raspotify or sound configuration ?

JasonLG1979, 2023-06-09:
No. It's related to your sound system configuration, ALSA or Pulseaudio. Librespot (the binary that the Raspotify service runs) output's stereo audio. Whatever you're outputting that to does whatever it does.

EG-Julien, 2023-06-09:
Thanks for the quick reply, I tried something found on ALSA / Jack documentation but It makes raspotify crash when I'm trying to connect to it.

Tried to add this in asound.conf

```conf
pcm.!jack {
    type jack
    # If "mpv --ao:alsa ..." gives you this error:
    # [ao/alsa] Unable to set buffer time near: Invalid argument
    # uncomment the following to allow non-jack-aligned period size:
    #align_psize = 0
    playback_ports {
        0 system:playback_2
        1 system:playback_3
    }
}

pcm.!default {
    type plug
    slave.pcm "jack"
    hint.description "Jack Audio"
}
````

JasonLG1979, 2023-06-09:
Well if you don't have jack installed that'll happen? I know nothing about jack. You don't need jack to mirror channels. [Here are the docs for configuring ALSA](https://vovkos.github.io/doxyrest/samples/alsa-sphinxdoc/page_pcm_plugins.html). It's a trial and error process, nothing I can really help you with unless I have the hardware in hand.