dtcooper/raspotify #632: Setting up dns-sd failed: Address already in use (os error 98)

apetrycki, 2023-06-17:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

I'm on Ubuntu 22.04 on x86.  I cannot figure out what the reason is for this error.  Nothing is running on 5353 (netstat and ss both come up clean).  I've tried removing the avahi-daemon package and rebooting with no luck.  I think this may have started after installing KVM and a virtual machine.  This used to work just fine.  I tried updating to the latest version of raspotify.  I've tried changing the discovery port in the config and through librespot command line.  I've seen that this issue may arise if wireguard is installed, but I don't have that installed on this system.  I've tried doing iptables -F to see if it might be a firewall issue.  I haven't been able to figure out where this error even comes from in the code to see if I can trace back what could be going wrong.

Any ideas on what could be causing this are welcome.

JasonLG1979, 2023-06-26:
This does not sound like a librespot/Raspotify issue. This sounds like a system configuration issue.

apetrycki, 2023-06-26:
I agree that I could likely fix it with some sort of system change, but the error message it throws seems incorrect.  It's difficult to troubleshoot without knowing what is actually wrong with the software.  Disabling discovery and manually entering a user and password in the configuration works, but then I assume only that account can use it.  Avahi has no problem running on that port, so why does librespot?  It seems to me, a change to librespot could prevent this issue and make it more stable, but it's hard to tell without knowing what the issue actually is.  Where does the error come from?  Is there a way to get a little more detail on what it's trying to do?

JasonLG1979, 2023-06-26:
Librespot uses it's own version of mDNS that is a best effort but not nearly as robust as Avahi in my experience. The best long term solution IMHO is for librespot to just use Avahi on Linux but someone would have to implement that.

JasonLG1979, 2023-06-26:
Avahi has a DBus interface that we could use pretty easily in theory. I'm pretty familiar with dealing with DBus interfaces in Python and JavaScript but I've never messed with them in Rust. It's on my long todo list but if anyone else has a more immediate need they are more then welcome to give it a shot.

apetrycki, 2023-06-26:
I've never coded in Rust before, but I can try taking a look at it.  I agree that using avahi is likely the best option since most systems come with it already installed from what I can tell.  I'll do some research on their dbus interface and Rust and see how big of a lift it'll be.

JasonLG1979, 2023-06-26:
Even if it's not currently installed as far as Raspotify as a .deb package is concerned it's simply a matter of making avahi a dependency.

closed 2023-10-02 by github-actions[bot]