dtcooper/raspotify #633: No 44.1khz sampling rate or S32 format on librespot -d ?

wyup, 2023-07-04:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

Can't see 44.1khz, S32 format and hw:CARD intead of dmix:CARD supported on

`librespot -d ?
[2023-07-04T12:40:51Z INFO  librespot] librespot 0.4.2 a6e1258 (Built on 2023-06-21, Build ID: g9ifMDjd, Profile: release)

        Compatible alsa device(s):

        --------------------------------------------------------------------

        Device:

                **dmix:CARD**=sndrpihifiberry,DEV=0

        Description:

                snd_rpi_hifiberry_digi, HiFiBerry Digi+ Pro HiFi wm8804-spdif-0
                Direct sample mixing device

        Supported Format & Sample Rate Combinations:

                Format: **S16**   Sample Rate(s): **48kHz**

        --------------------------------------------------------------------`

On /etc/raspotify/conf, i've set:

```
LIBRESPOT_FORMAT="S32"
LIBRESPOT_BACKEND="alsa"
LIBRESPOT_DEVICE="hw:CARD=sndrpihifiberry"
LIBRESPOT_INITIAL_VOLUME="100"
LIBRESPOT_NORMALISATION_PREGAIN="-10.0"
```
aplay -L is:

```
root@DietPi:/etc/raspotify# aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
**hw:CARD=sndrpihifiberry**,DEV=0
    snd_rpi_hifiberry_digi, HiFiBerry Digi+ Pro HiFi wm8804-spdif-0
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_digi, HiFiBerry Digi+ Pro HiFi wm8804-spdif-0
    Hardware device with all software conversions
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_digi, HiFiBerry Digi+ Pro HiFi wm8804-spdif-0
    Default Audio Device
dmix:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_digi, HiFiBerry Digi+ Pro HiFi wm8804-spdif-0
    Direct sample mixing device
```
amixer:
```
root@DietPi:/etc/raspotify# amixer
Simple mixer control 'Tx Source',0
  Capabilities: enum
  Items: 'S/PDIF RX' 'AIF'
  Item0: 'AIF'
```

My audio card is a SPDIF digital transport compatible with hifiberry-digi-pro, displayed as:
hw:0,0 : sndrpihifiberry HiFiBerry Digi+ Pro HiFi wm8804-spdif-0
(it's dtoverlay is not necessary entered in /boot/config.txt)

My interest is to support native 44.1khz on raspotify and select, if possible hw:0,0 instead of dmix

DietPi v8.19.1 : 12:52 - mar 04/07/23 (debian)
Device model : RPi 4 Model B (aarch64)
6.1.21-v8+ kernel

nothing relevant on journalctl -u raspotify -b

thanks for support

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
jul 04 13:42:05 DietPi systemd[1]: Stopped raspotify.service - Raspotify (Spotify Connect Client).
jul 04 14:40:48 DietPi systemd[1]: Started raspotify.service - Raspotify (Spotify Connect Client).
jul 04 14:40:49 DietPi librespot[4477]: [2023-07-04T12:40:49Z WARN  librespot] Without the `--enable-volume-normalisation` / `-N` flag normalisation options have no effect.
```

JasonLG1979, 2023-07-05:
Is it muted?

wyup, 2023-07-16:
I hadn't tested it with music playback, just 'librespot -d ?' output idle.
Now it's working ok, according to my dac, input is 44.1Khz.

This what 'librespot -d ? says:
`root@DietPi:~# librespot -d ?
[2023-07-16T17:05:38Z INFO  librespot] librespot 0.4.2 a6e1258 (Built on 2023-06-21, Build ID: g9ifMDjd, Profile: release)

        Compatible alsa device(s):

        --------------------------------------------------------------------

ALSA lib pcm_dmix.c:999:(snd_pcm_dmix_open) unable to open slave`

JasonLG1979, 2023-07-16:
You can't prove devices while they are in use.

closed 2023-09-07 by wyup