dtcooper/raspotify #635: Raspotify, Music History

TruePcKing, 2023-07-12:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Not sure if this has been noticed before, and this is probably more of a Spotify issue, but music played from the Raspotify aren't categorized under my playlist. 

You can view music history on the mobile app
[image]

JasonLG1979, 2023-07-12:
That is a know issue with librespot.

JasonLG1979, 2023-07-12:
It's related to this:

https://github.com/librespot-org/librespot/issues/1127

Librespot doesn't report back to Spotify that it's playing or has player a track.

closed 2023-07-12 by JasonLG1979