dtcooper/raspotify #636: My phone disconnects from raspberry stream

xfirebg, 2023-07-13:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

Hello, mi phone disconnects from my Pi straming service. Here is the log from the device.
After that I need to connect again with my phone.

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
Jul 13 14:10:24 raspberrypi librespot[1744]: [2023-07-13T11:10:24Z ERROR librespot_playback::player] Unable to load audio item: MercuryError
Jul 13 14:10:24 raspberrypi librespot[1744]: [2023-07-13T11:10:24Z ERROR librespot_core::session] Connection reset by peer (os error 104)
Jul 13 14:10:24 raspberrypi librespot[1744]: [2023-07-13T11:10:24Z ERROR librespot_connect::spirc] subscription terminated
Jul 13 14:10:24 raspberrypi librespot[1744]: [2023-07-13T11:10:24Z WARN  librespot] Spirc shut down unexpectedly
Jul 13 14:10:24 raspberrypi librespot[1744]: [2023-07-13T11:10:24Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gue1.spotify.com:4070
```

JasonLG1979, 2023-07-13:
See the upstream issue:

https://github.com/librespot-org/librespot/issues/1151

closed 2023-07-13 by JasonLG1979

xfirebg, 2023-07-14:
I see that there is a fix to this in this topic: https://github.com/librespot-org/librespot/pull/1129
How to implement this in Raspotify?

xfirebg, 2023-07-17:
> That's not a fix for 104's.

Strange, because there are 2 mentions in your link provided on the top. One is yours, the [other one](https://github.com/hrkfdn/ncspot/issues/1054) is of zborro. There I tracked that he explain for the 104's. So you need to told him that this is not the same issue too.

JasonLG1979, 2023-07-17:
A fix to 104's would be reconnecting seamlessly.