dtcooper/raspotify #641: Everything seems fine but no sound (unable to create IPC semaphore)

emc02, 2023-08-28:
### Compatible Device

- [X] I'm running Raspotify on a compatible Device

### Are you sure?

- [X] I'm not trying to run Raspotify on a ARMv6 device

### Compatible OS

- [X] I'm running Raspotify on a compatible OS

### Compatible Configuration

- [X] I'm running Raspotify on a compatible Configuration

### Latest Version

- [X] I'm running the latest version of Raspotify

### Are you sure?

- [X] I'm not running Raspotify 0.31.8.1 on a Pi v1 or Pi Zero

### Due Diligence

- [X] I have done my due diligence

### What happened?

Setup:
Raspberry Pi3
USB Sound card

Had a working setup (with a raspotify version from begin of 2023 (do not know which one unfortunately)) and did a "apt full-upgrade".
Suddenly there was no spotify sound anymore.

speaker-test -c 2 gives the pink noise on both speakers

during "playback" librespot -d ? shows:
ALSA lib pcm_dmix.c:1038:(snd_pcm_dmix_open) unable to create IPC semaphore

changing the alsa.conf from
defaults.pcm.ipc_perm 0660
to
defaults.pcm.ipc_perm 0666

avoids the error, but it's not playing anyways.

without playback librespot -d ? shows:
`[2023-08-28T10:56:05Z INFO  librespot] librespot 0.4.2 a6e1258 (Built on 2023-06-21, Build ID: zuwVQKTQ, Profile: release)

        Compatible alsa device(s):

        --------------------------------------------------------------------

        Device:

                iec958:CARD=ICUSBAUDIO7D,DEV=0

        Description:

                ICUSBAUDIO7D, USB Audio
                IEC958 (S/PDIF) Digital Audio Output

        Supported Format & Sample Rate Combinations:

                Format: S16   Sample Rate(s): 44.1kHz, 48kHz, 96kHz

        --------------------------------------------------------------------

        Device:

                dmix:CARD=ICUSBAUDIO7D,DEV=0

        Description:

                ICUSBAUDIO7D, USB Audio
                Direct sample mixing device

        Supported Format & Sample Rate Combinations:

                Format: S16   Sample Rate(s): 48kHz

        --------------------------------------------------------------------
`

The librespot log shows the songs and that it's playing through alsasink.

Tried also another USB sound card without success...

Please help, thanks!

### Relevant log output and/or the contents of /etc/raspotify/crash_report if any ( sudo journalctl -u raspotify -b and sudo cat /etc/raspotify/crash_report )

```shell
pi@soundsystem:~ $ sudo journalctl -u raspotify -b
-- Journal begins at Thu 2022-09-22 01:23:40 BST, ends at Mon 2023-08-28 11:58:11 BST. --
Aug 28 11:55:10 soundsystem systemd[1]: Started Raspotify (Spotify Connect Client).
Aug 28 11:57:38 soundsystem librespot[444]: [2023-08-28T10:57:38Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:4070
Aug 28 11:57:38 soundsystem librespot[444]: [2023-08-28T10:57:38Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:443
Aug 28 11:57:38 soundsystem librespot[444]: [2023-08-28T10:57:38Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gew4.spotify.com:80
Aug 28 11:57:38 soundsystem librespot[444]: [2023-08-28T10:57:38Z WARN  librespot_core::apresolve] Ignoring blacklisted access point ap-gue1.spotify.com:80
```

emc02, 2023-08-28:
I even tried an older raspberryos image with no luck

emc02, 2023-09-01:
Ways to reproduce (used Raspberry Pi3 Model B+):
Flash latest (bullseye) RaspberryOS lite 32bit on SD Card
add these lines to config.txt:
gpu_mem=16
dtparam=audio=off

boot
sudo apt update
sudo apt full-upgrade

sudo nano /usr/share/alsa/alsa.conf
set:
defaults.ctl.card 1
defaults.pcm.card 1

sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh

result: no sound
after reboot: no sound

librespot -d ?
ALSA lib pcm_dmix.c:1038:(snd_pcm_dmix_open) unable to create IPC semaphore

Please help...

emc02, 2023-09-01:
ok, have the solution: my sound card suddenly stops supporting 44,1kHz (new driver with update?) so I had to set:
LIBRESPOT_SAMPLE_RATE="48kHz"
LIBRESPOT_INTERPOLATION_QUALITY="High"

and now everything works well again!

emc02, 2023-09-01:
and now everything works well again!

closed 2023-09-01 by emc02

pReya, 2023-10-17:
Thanks so much @emc02 I had the exact same problem!