dtcooper/raspotify #642: Recommended Specs

darthmorf, 2023-08-30:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi,

Are there any minimum recommended specs for running raspotify? I'm looking to purchase a new Pi now they're back in stock for this as all my existing ones are deployed so figured I'd check what the requirements are. I was planning on using a Zero 2 W along with an audio board (e.g. https://shop.pimoroni.com/products/pirate-audio-line-out?variant=31189750546515). Does this seem suitable?

I've searched around for this a little but I couldn't find much other than the fact that Zero 1.x is no longer supported. 

Cheers

JasonLG1979, 2023-08-31:
For sure, a Pi Zero v2 will work great. I have one and run the 64 bit version of Raspberry Pi OS Lite (the headless version, no Desktop Environment) on it. Raspotify runs great on it. That's basically my reference potato. I make sure Everything works really well on the PI Zero v2. I use a USB DAC (Topping D10S) I've never used that HAT so I couldn't tell you anything about the audio quality? It probability uses the same basic design as just about every other PI HAT DAC? I'm assuming it's meh, in the grand scheme of things but you gotta consider the price, which ain't bad. I also don't know anything about the screen or their software to display and control(?) Spotify? Raspotify doesn't have anything to do with any of that. I personally would up my budget to like $60 and get a better USB dongle DAC.

darthmorf, 2023-08-31:
> For sure, a Pi Zero v2 will work great. I have one and run the 64 bit version of Raspberry Pi OS Lite (the headless version, no Desktop Environment) on it. Raspotify runs great on it. That's [...]

Thanks - glad to hear the zero 2 is ideal for this. I'll look into USB DACs instead, I assume basically anything will work?

Also; do you mind if I submit a PR to the FAQ to mention that the Zero 2 or better will work well?

emc02, 2023-09-01:
@darthmorf please report back if a fresh installation is working for you (I had serious troubles with latest Bullseye RaspberryOS...) (see my issue opened here)

JasonLG1979, 2023-09-01:
> Also; do you mind if I submit a PR to the FAQ to mention that the Zero 2 or better will work well?

Raspotify will run fine on all Pi's except Pi 1's Pi Zero v1's and the 1st revision of the Pi 2. Basically any Pi that doesn't have an ARMv6 CPU. The install script checks to make sure the device is compatible. I don't think it's really needed?

closed 2023-12-09 by github-actions[bot]