dtcooper/raspotify #650: Can't install Raspotify (see inside for error message)

heinzgruber, 2024-02-15:
### Due Diligence

- [X] I have done my due diligence

### What can we help you with?

Hi,
my raspotify installation was broken and i've ha dno idea why. uninstalled it via 

sudo apt-get remove -y raspotify

and deleted the rapsotify folder in /etc

when i try to reinstall via 

sudo apt-get -y install curl && curl -sL https://dtcooper.github.io/raspotify/install.sh | sh

nothing happens and i get this output:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
curl is already the newest version (7.74.0-1.3+deb11u11).
The following packages were automatically installed and are no longer required:
  libasyncns0 libopus0 libpulse0 libsndfile1 libvorbisenc2 raspinfo
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

What am i doing wrong?

Thanks

mquevill, 2024-05-01:
When I tried to replicate your steps, the new `/etc/raspotify` directory seemed to have restrictive permissions:
```
drw-------   2 root root    4096 May  1 15:25 raspotify
```
and
```
-rw-------   1 root root  5015 Jun 21  2023 conf
```
Furthermore, I then get issues listed in `sudo journalctl -u raspotify -b` that specify `librespot: Failed to get home directory.`

This is with a Raspberry Pi OS Desktop 12 on a Raspberry Pi 2 (32 bit). This definitely seems like strange behavior on a reinstall.

mquevill, 2024-05-02:
I retried this process with a fresh install of Raspberry Pi OS Lite 12, and it was able to reinstall properly. The permissions are how the files appear even on first install; they just seem a bit restrictive compared to other directories.

@heinzgruber You may need to debug the install.sh script on your machine to pinpoint where it's going wrong. The script is set up to have error checking + error messages, but it's hard to tell what's happening with the limited output you included.

closed 2024-08-09 by github-actions[bot]